Jomelo/LCDMenuLib2 #3: Single menu option

paulosilver, 2017-11-01:
Hello, i found that one menu single option does not work

This works
LCDML_add         (0  , LCDML_0         , 1  , "OP1"   , NULL);
  LCDML_add         (1  , LCDML_0_1       , 1  , "1"    , NULL);                   
  LCDML_add         (2  , LCDML_0_1       , 2  , "2"    , NULL);                   
  LCDML_add         (3  , LCDML_0_1       , 3  , "3"             , NULL);
  LCDML_add         (4  , LCDML_0         , 2  , "OP2"           , NULL);

This does not work
LCDML_add         (0  , LCDML_0         , 1  , "OP1"   , NULL);
  LCDML_add         (1  , LCDML_0_1       , 1  , "1"    , NULL);                   
  LCDML_add         (2  , LCDML_0_1       , 2  , "2"    , NULL);                   
  LCDML_add         (3  , LCDML_0_1       , 3  , "3"             , NULL);

Jomelo, 2017-11-01:
I can reproduce it and search the bug now :-)

Jomelo, 2017-11-01:
Its fixed: 
https://github.com/Jomelo/LCDMenuLib2/releases/tag/v1.0.1

closed 2017-11-01 by Jomelo