Jomelo/LCDMenuLib2 #5: Please support ESP32!

tackelua, 2018-01-03:
I test with LCD_I2C
Compile with ESP32 Dev Kit and it's error
`LiquidCrystal.cpp: 190:48: error: 'analogWrite' was not declared in this scope
   analogWrite ( _backlightPin, value )`
Compiler can not found <Arduino.h> in file LiquidCrystal.cpp
`#if (ARDUINO <  100)
#include <WProgram.h>
#else
#include <Arduino.h>
#endif`
But the problem, we use LCD I2C it's mean LiquidCrystal_I2C.h but compiller also use LiquidCrystal.h

Jomelo, 2018-01-03:
Hello, i test the library with an ESP32 and it works. 

I think you have a problem with your I2C library.  I think you have to find a new I2C library which works with an ESP32.

Jomelo, 2018-01-03:
I remove one bug with ESP8266 and ESP32 in version 1.0.2. But i cannot find a bug in the liquidCrystal lib. This lib is written from other programers.

tackelua, 2018-01-04:
>  I think you have to find a new I2C library which works with an ESP32.

Can you give me a link of the library that you using? Pls
Thank you!

Jomelo, 2018-05-02:
I am currently not using any i2c displys.

closed 2018-05-02 by Jomelo