Jomelo/LCDMenuLib2 #18: Unsure how to integrate a dynamic submenu with U8g2

RawLiquid, 2018-04-10:
Hi,
This isn't so much of an issue with the library as it is me being unable to figure out how to go about adding a dynamically generated submenu to a U8g2 sketch.
I've managed to create an extensive menu, and up till this have managed to get it to at least mostly do what I want.

I have a variable named filelist[][20] which expands based on how many of a particular structure are written in the EEPROM, and I've parsed out the Name variable, and copied each into filelist[]. currently there are 9 items.

filelist[0]="BBS 01"
filelist[1]="File 02"
and so on.

I managed to put them in by adding some dummy menu items, and displaying a custom string when the getid() matches, however that obviously won't work once the user either deletes an item, or adds a new one....

Could anybody point me in the right  direction here?

RawLiquid

Jomelo, 2018-04-10:
Hello, 
it is an interesting thing/problem. 

I make an example for your based on the serial monitor example. 
You can download the example here:
[](https://jomelo.de/r/arduino/LCDML_900_dynamicStaticList.zip)

I use to generate the dynamic function a "dirty" makro because i would not copy the function 20 times :-)) . You can replace the makro with your code or you can modify it. 

Regards
Nils F.

RawLiquid, 2018-04-11:
Thank you so much for taking the time to do this. Unfortunately I am not able to release the code I'm working on, mostly thanks to search engines,  but I do have it stored in a private repo here. If you would  like to see  the currently rather messy work in progress, along with the changes I made(particularly to the control module), I can add you as a user to the repo....
Who knows, you might take one look at it and notice that I've gone about doing something in a fundamentally flawed way....
Just say the word and I will send the invite...Now, to go and study this code you provided and see if I can't make the dynamic menu happen....
RawLiquid

RawLiquid, 2018-04-11:
Interesting... So your basically saying, rather than spend time trying to create the submenu on the fly, add a shell entry for each of the possible storage slots, and hide the inactive entries using the condition functions, and use the normal Dynamic text to fill in the entries on the fly....

So, the next question, is it possible to 'reuse' a submenu without having to re-enter it for each item?
for example, the (formerly) dynamic submenu was going to be 'Profiles', listing the 20 entries. pressing 'enter' on any of those entries, then gives you the following choices...
1) Load (calls a function to retrieve the selected profiles stored settings from EEPROM
2) Overwrite (calls a function that saves a list of settings currently in memory to EEPROM, in the slot that was selected
3) Rename (allows the user to change the 'name' data for the selected slot
4) Delete (updates an inUse flag from 1 to 0 in eeprom, removing the profile from the menu's, and (eventually) allowing the data to be overwritten by a 'new' profile being saved

On a separate note, What is the purpose of specifying the new_num parameter?
I had hoped that it was used  for sorting the items in a submenu,
for example if you defined the menu items, then needed to add an item to the top of  the menu without having to go back and re-number things....Obviously that didn't work out, the menu was sorted by the main ID...

Jomelo, 2018-04-11:
Hello, 
the parameter can only be used in a normal menu function. In a dynamic menu function the line where the element is displayed is the parameter.

I have add new example code:
[Download](http://jomelo.de/r/arduino/LCDML_900_dynamicStaticList_2.zip)

The code shows you, how you can change the cursor position and how parameters are used in this function. 

With "jump" and "setCursorTo" function you can go into hidden menus. 

A sort function for submenus is not included in this lib. 
When you would sort element ids you can create a secound index list with 20 elements of a uint_8t datatype. Now you can sort the index list and edit the makro function to call the following code:

Serial.print(filelist[index_list[ELEMENT_ID]]);\

RawLiquid, 2018-04-13:
I think I've just hit some kind of  limit that isn't obvious....Suddenly the first few menu items disappeared, then I added another sub item, and another item from the beginning disappeared....

RawLiquid, 2018-04-13:
[image]

RawLiquid, 2018-04-14:
btw, I'm currently at 114 elements, with 3 missing, making the limit 111??? seems kinda arbitrary to me

Jomelo, 2018-04-15:
Without locking into your code i cannot help. 
Can you invite me to your repo ?

RawLiquid, 2018-04-15:
I sent it sbout 4 hours ago or so, if you havent seen it check spam....

RawLiquid, 2018-04-27:
Don't suppose you had any thoughts on the menu items disappearing?

closed 2018-05-02 by Jomelo