Jomelo/LCDMenuLib2 #21: control rollover using rotary encoder

RawLiquid, 2018-05-12:
I disabled the rollover, but I've noticed that going one direction it stops as expected, but going the other direction continues to roll over, making selecting upper values tricky since if you go one too far, your at the bottom again...

Jomelo, 2018-05-12:
I have reproduced the problem and upload a bugfix. 
The menu is not moving to far, but with v1.2.5 i change the display clear handling and forgott to update the cursor position. 
The bugfix is online (master branch) but i have not created a new version because i will make some tests tomorrow.

RawLiquid, 2018-05-12:
No problem, I’m curious to find out where you fixed it as I had started to
try and find the issue, and failed. I will admit to not spending much time
on it though. Thanks for the quick fix.

closed 2018-12-10 by Jomelo