Jomelo/LCDMenuLib2 #22: MENU_getLayer on Root menu

mavrickJ, 2018-06-17:
Hello
I encounter a small problem: how do I know that we are on the root menu?

the LCDML.MENU_getLayer () function returns 0 as the first sub-menu
same for getID ()
and more generally the same for all the functions returning to a notion of position in the menus.

I thought, wrong that LCDML.MENU_getLayer () would return 255!

What solution is there?

Thank you in advance
Best regards

Ps: sorry for my poor english

Jomelo, 2018-06-17:
Hello, 
i have no problem with your english, my is not better :-)

The displayed menu is every a submenu. The main root element cannot be called or opened. The element with the id 255 is the root element. 

The first submenu is on layer 0 (main displayed menu)

At the moment there is no direct call to check the root menu. All elements on layer 0 are on the first submenu .

The first element can be checked with:

```
if(LCDML.MENU_getLayer() == 0 && LCDML.MENU_getCursorPos() == 0 && LCDML.MENU_getCursorPosAbs() == 0 && LCDML.MENU_getScroll() == 0) 
{
 // ...
}
```

mavrickJ, 2018-06-18:
Hello and thank you

In fact I will need to execute a piece of code, in the LCDML_control loop, only when one is on the root main menu, whatever the position of the cursor.

I looked a little and would it be possible in MENU_getParentId () to return a particular value for the root of the menu?
of style
uint8_t LCDMenuLib2 :: MENU_getParentId ()
/ * ************************************************ ******************** * /
{
     if (layer> 0 && layer <_LCDML_DISP_cfg_cursor_deep) {
         return _LCDML_NO_FUNC;
     }
else return 255;
}

It would not take all the cases but it would know that there is no parent level.
What do you think about?

Best regards

RawLiquid, 2018-06-18:
What he is trying to say is, using a menu function executes on every
refresh, it’s the if statements that prevent things from happening....

mavrickJ, 2018-06-25:
Hello,
Thank you.
finally i get around the difficulty by creating a global variable that i assign in each procedure and exit a boolean procedure.
in the rest of the arborescence the problem does not arise, since I can use the function LCDML.MENU_getLayer ().

Thank you
best regards