Jomelo/LCDMenuLib2 #23: Dynamic submenu for listing cd cards

Tigeerg, 2018-06-19:
I'm sorry, I did not find an example or a mention in the code of how to implement a dynamic submenu. Saw mention of this issue in this thread, https://github.com/Jomelo/LCDMenuLib2/issues/18, but the file you attached is no longer available.
Also found a function in the previous version of the LCDMenuLib library: https://github.com/Jomelo/LCDMenuLib/issues/28.
This functionality in LCDMenuLib2 is not implemented?

Jomelo, 2018-07-05:
Hello,
i have not implemented the example to LCDMenuLib2 but i can do on the next days. :-)

e1z0, 2021-09-26:
It does not work with u8g2 lib, only serial output is visible but dynamic menu not shown on the screen. I'm seeing that i can navigate and it shows on the serial.

Jomelo, 2021-09-26:
Hello, 
not all examples support the complete functionality. You have to import the right content from one example to another one. Then it works. 

The difficult on the example with the filelist ist not the display output.