Jomelo/LCDMenuLib2 #30: open and close the entire menu at will?

RawLiquid, 2018-10-05:
Might there be any way to say have a sketch running that uses the lcd, then when the user presses a button show the menu and at some point be able to exit the menu and go back to the initial program?

Zefram88, 2018-12-09:
I've managed to do this using one of the menus as main for program,  i've call it directly in void setup() as this:
`  LCDML.OTHER_jumpToFunc(mFunc_information);
`so when it boot up it display directly this page, plus i've used 
```
    if (LCDML.BT_checkEnter()) { // check if any button is pressed (enter, up, down, left, right)
      // LCDML_goToMenu stops a running menu function and goes to the menu
      LCDML.FUNC_goBackToMenu();
```
so it goes to the menu only if you press enter.

Jomelo, 2018-12-10:
Its a bug i am working on it.

closed 2019-01-13 by Jomelo