Jomelo/LCDMenuLib2 #31: Screensaver does not work in i2c LCD example code when menu entry is removed

SeeJayDee, 2018-10-30:
In the example sketch `LCDML_031_i2c_display_20x4` the screensaver function only works when bound to a menu entry.

At line 90 of `LCDML_031_i2c_display_20x4.ino` I changed:
`LCDML_add(17, LCDML_0, 6, "Screensaver", mFunc_screensaver);`
to:
`LCDML_add(17, LCDML_0, 6, "Screensaver", NULL);`
Everything else in the sketch was left as-is.

As a result, the screensaver function never runs despite the activation time having elapsed.
`LCDML.SCREEN_enable(...)` was called properly in `setup()`.
<br><br><br>

I tried to diagnose the cause of this myself but got lost amongst all the macros. >.>
<br>
Hardware: Arduino Uno R3, generic HD44780 display connected over i2c

Jomelo, 2018-11-01:
Hello, 
i will check this tomorrow.

Jomelo, 2018-12-10:
I fixed this bug in Version 1.3.0

closed 2018-12-10 by Jomelo