Jomelo/LCDMenuLib2 #32: Bug in all examples with ir remote control

Jomelo, 2018-11-14:
The message they sent you was:

thanks for your great work
trying your examples
i had found one bug later
in control section for irremote please add this line for correct act for IRremte 
>>
switch (key)
    {
      case '#': LCDML.BT_enter(); break;
      case '2': LCDML.BT_up();    break;
      case '8': LCDML.BT_down();  break;
      case '4': LCDML.BT_left();  break;
      case '6': LCDML.BT_right(); break;
      case '*': LCDML.BT_quit();  break;
      default: break;
    }
    irrecv.resume(); // Receive the next value

clear-sky, 2018-11-18:
if add this line after switch function every thing works ok!
irrecv.resume();

Jomelo, 2018-12-10:
i add this bugfix to the develop branch

closed 2018-12-10 by Jomelo