Jomelo/LCDMenuLib2 #35: screen saver opens menu if dyn content is visible

eimix, 2018-12-02:
In example oginal "LCDML_021_dynUpdatedContent"
(changed to work to 16x2 LCD)

Made 2 videos (attached zip, video is not possible to attach):
[video_menu_lcd16x2.zip](https://github.com/Jomelo/LCDMenuLib2/files/2637185/video_menu_lcd16x2.zip)

First one  with no dynamic content visible - after 10 seconds screensaver activates and stays visible.
Second - screensaver activates and instantly disapears (no button is presed). It looks like that screen saver activations works like ENTER button. (this is problem).

Tried other menu tree - with dynamic content and sublevels
1
2 
3 dynamic
  3.1
  3.2
  3.3
4

If 3rd item is visible and has cursor on it
 - after 10 seconds 3.1 3.2 3.3 is displayed but not screensaver.

Jomelo, 2018-12-02:
Hello, 
Version 1.2.9 is not the best version there are some bugs with the screensaver and the "jumpTo" xx function. 
At the moment i am working on an update. The news beta version is under the develop branch:
https://github.com/Jomelo/LCDMenuLib2/tree/develop

eimix, 2018-12-02:
Something is still unexpected
this is my menu sample
`
*********************************************************************************
// LCDML_add(id, prev_layer, new_num, lang_char_array, callback_function)
  LCDML_add         (0  , LCDML_0         , 1  , "Status"         , mFunc_screensaver);

  LCDML_add         (1  , LCDML_0         , 2  , "Whats up?"        , NULL); 
  LCDML_add         (2  , LCDML_0_2       , 1  , "A"                , mFunc_back); //TODO
  LCDML_add         (3  , LCDML_0_2       , 2  , "B"                , mFunc_back); //TODO 
  LCDML_add         (4  , LCDML_0_2       , 3  , "C"                , mFunc_back); //TODO
  LCDML_add         (5  , LCDML_0_2       , 4  , "Back"             , mFunc_back);       
   
  LCDML_add         (6  , LCDML_0         , 3  , "Stage 1"          , NULL); 
  LCDML_add         (7  , LCDML_0_3       , 1  , "A"                , mFunc_back); //TODO
  LCDML_add         (8  , LCDML_0_3       , 2  , "B"                , mFunc_back); //TODO
  LCDML_add         (9  , LCDML_0_3       , 3  , "C"                , mFunc_back); //TODO
  LCDML_add         (10  , LCDML_0_3      , 4  , "D"                , mFunc_back); //TODO
  LCDML_add         (11  , LCDML_0_3      , 5  , "E"                , mFunc_back); //TODO
  LCDML_add         (12 , LCDML_0_3       , 6  , "Back"             , mFunc_back);       

  LCDML_add         (13 , LCDML_0         , 4  , "Stage 2"         , NULL); 
  LCDML_addAdvanced (14 , LCDML_0_4       , 1  , NULL,              "S2"                , mDyn_para,               0,   _LCDML_TYPE_dynParam); //TODO
  LCDML_addAdvanced (15 , LCDML_0_4       , 2  , NULL,              "T2"                , mDyn_para,               0,   _LCDML_TYPE_dynParam); //TODO
  LCDML_addAdvanced (16 , LCDML_0_4       , 3  , NULL,              "H2"                , mDyn_para,               0,   _LCDML_TYPE_dynParam); //TODO
  LCDML_addAdvanced (17 , LCDML_0_4       , 4  , NULL,              "L2"                , mDyn_para,               0,   _LCDML_TYPE_dynParam); //TODO
  LCDML_addAdvanced (18 , LCDML_0_4       , 5  , NULL,              "Start 2"           , mFunc_p2,                2,    _LCDML_TYPE_default);
  LCDML_add         (19 , LCDML_0_4       , 6  , "Back"            , mFunc_back);       

  LCDML_add         (20 , LCDML_0         , 5  , "Settings"       , NULL); 
  
  // ***TIP*** Try to update _LCDML_DISP_cnt when you add a menu element.

  // menu element count - last element id
  // this value must be the same as the last menu element
  #define _LCDML_DISP_cnt    20
*************************************************************************************************
`

putted it in example LCDML_021_dynUpdatedContent

if menu item selected (cursor on it, but not entered):
"Status" - screensaver activated after 10s
"Whats up?" and "Stage 1" - after 10s screen redraw, menu still shown
 "Stage 2" - after 10 seconds - menu is entered and sub level is displayed

I could make more videos if needed.

I got messed up, and unable to debug it, please help.

Jomelo, 2019-01-13:
Hello, 
Can you test this version: https://github.com/Jomelo/LCDMenuLib2/tree/develop

I have rewrite the complete library code because there were some bugs which i could not fix on the last code base.

One Line in the LCDML_display_menu tab have been changed.
old:
if ((tmp = LCDML.MENU_getObj()->getChild(LCDML.MENU_getScroll()))

new: 
if ((tmp = LCDML.MENU_getDisplayedObj()) != NULL)

closed 2019-02-28 by Jomelo