Jomelo/LCDMenuLib2 #36: Set range in LCDML_addAdvanced

Zefram88, 2018-12-12:
As title, is it possible to set a range different than 0-255 in `LCDML_addAdvanced`?

closed 2019-01-13 by Jomelo

Scabattoir, 2020-06-08:
Hi,
my question would connect to this topic:
is there a way to make it floating point with a fixed range?
Let's say I'd want 0.0 to 5.0 in 0.1 steps.
Is it possible?

Jomelo, 2020-06-08:
Hello, 
floating point values are not supported. 

Would you use this value as an parameter or for a dynamic menu element? 

Parameter: it is not supported but divide your parameter value through 10. (Range 0 - 50/10)
Dynamic menu element: you can change the output to your datatype you need.

Scabattoir, 2020-06-08:
I'd set a variable with it.
For me it is okay just to have it on the display like uint8_t/10.0 and set the variable accordingly but I can't make it show on the screen as the display format is locked.
It is the dynParam type I'm talking about.