Jomelo/LCDMenuLib2 #37: Variable was not declared in LCDMenuLib2_macros.h

sakunamary, 2018-12-28:
I am trying to compile the examples of U8g2 menu  in Sloeber,the Eclipse IDE for arduino , but not working .
here is the error output :

15:34:03 **** Incremental Build of configuration Release for project menu2 ****
"E:\\sloeber\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ..\sloeber.ino.cpp'
'Starting C++ compile'
"E:\sloeber\arduinoPlugin\packages\arduino\tools\avr-gcc\5.4.0-atmel3.6.1-arduino2/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR     -I"E:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.23\cores\arduino" -I"E:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.23\variants\mega" -I"E:\sloeber\arduinoPlugin\libraries\LCDMenuLib2\1.3.0\src" -I"E:\sloeber\arduinoPlugin\libraries\U8g2\2.24.3\src" -I"E:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.23\libraries\SPI\src" -I"E:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.23\libraries\Wire\src" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\sloeber.ino.cpp"  -o  "sloeber.ino.cpp.o"

In file included from E:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.23\cores\arduino/Arduino.h:28:0,
                 from ..\sloeber.ino.cpp:7:
..\LCDML_display_menu.ino: In function 'void lcdml_menu_display()':
E:\sloeber\arduinoPlugin\libraries\LCDMenuLib2\1.3.0\src/LCDMenuLib2_macros.h:66:48: error: **'g_LCDML_DISP_lang_table' was not declared in this scope**  --I think here is the key point.... I have checked the code "LCDMenuLib2_macros.h".
          strcpy_P(var, (char *)pgm_read_word(&(g_LCDML_DISP_lang_table[id]))); \
                                                ^
..\LCDML_display_menu.ino:66:13: note: in expansion of macro 'LCDML_getContent'
             LCDML_getContent(content_text, tmp->getID());
             ^
In file included from ..\sloeber.ino.cpp:98:0:
..\LCDML_display_menu.ino:35:11: warning: unused variable 'scrollbar_min' [-Wunused-variable]
   uint8_t scrollbar_min     = 0;
           ^
..\LCDML_display_menu.ino:38:11: warning: unused variable 'scroll_pos' [-Wunused-variable]
   uint8_t scroll_pos        = ((1.*n_max * _LCDML_DISP_rows) / (scrollbar_max - 1) * scrollbar_cur_pos);
           ^
subdir.mk:18: recipe for target 'sloeber.ino.cpp.o' failed
make: *** [sloeber.ino.cpp.o] Error 1

15:34:07 Build Failed. 3 errors, 9 warnings. (took 3s.759ms)

I have duble checked in Arduino IDE ,the examples project is working ... so could you pls have a look on this issue ...Thanks a lot

Jomelo, 2019-01-14:
Hello, 
i have not used this IDE. It seems that the order of the header files is incorrect or a header file is not included in any file.

chrismiller, 2019-02-05:
Hi @sakunamary, did you figure out a solution to this? I'm seeing the same problem using PlatformIO + CLion as my IDE (compiling with CMake under the hood).

YannikW, 2019-04-28:
I'm getting the same error with PlatformIO - is there any solution?

velvettiere, 2020-01-06:
I have The seams problem with visual studio code. The program LCDMENULIB2 work perfectly with Arduino code IDE But this much errors Came in compilation process on visual studio code. I'm Very dislike :(. I Thing that The error are in LinkedIn library But no understood like sole that. Can you help me? Thanks a Lot.

velvettiere, 2020-01-07:
I have seen that This code have more problems with c/c++ sintax And deployment. I have resolve partial error But Now remain more problem with lcdmenulib-macros.h in macro table definition :(

sdp8483, 2020-01-19:
I am able to compile this library using PlatformIO for Teensy LC.
I placed most of the functions from the beginner LCD example into separate `.cpp` files along with appropriate header files, for example `LCDML_condition.ino` became `LCDML_condition.cpp` along with `LCDML_condition.h` These files are then placed in `/lib/LCDML` and the header files are included in `main.cpp`

The only issue I ran into was as describe above. To solve that I found I had to place the `void lcdml_menu_display()` function in `main.cpp`

jdevine82, 2020-02-14:
Actually I am using the Arduino IDE and when modifiying the example even the slightest I get the above problem.

jdevine82, 2020-02-14:
OK i have found my problem and it may be yours too as my error was exactly the same. The line under the menu definition
 #define _LCDML_DISP_cnt    13  // This must be the number of menu items.

prathameshjakka, 2020-05-21:
> I am able to compile this library using PlatformIO for Teensy LC.
> I placed most of the functions from the beginner LCD example into separate `.cpp` files along with appropriate header files, for [...]

Hey, can you upload the files with the changes you have made? I'm having the same problem and not able to fix it.

aerickson, 2020-06-03:
I solved this by just combining the contents of all of the .ino files into a single .cpp file and then creating a header file. 

For the header file I used a program called `makeheaders` (https://www.hwaci.com/sw/mkhdr/) and then commented out everything that wasn't a function declaration (or you can comment everything out, compile, enable the lines to fix the first error, repeat until linking).

acitonic, 2021-04-17:
I have the same issue ....
I don't know how to solve this Problem
both terms use things which are declared later in the other term

**Here my code:**
<pre>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><b><font color="#d35400">LCDMenuLib2</font></b><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

 &nbsp;<font color="#5e6d03">#define</font> <font color="#00979c">_LCDML_DISP_cfg_cursor</font> &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#000000">0x7E</font> &nbsp;&nbsp;<font color="#434f54">&#47;&#47; cursor Symbol</font>
 &nbsp;<font color="#5e6d03">#define</font> <font color="#00979c">_LCDML_DISP_cfg_scrollbar</font> &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#000000">1</font> &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#434f54">&#47;&#47; enable a scrollbar</font>
 &nbsp;<font color="#434f54">&#47;&#47; enable this line (set to 1) to show a header above the first menu element</font>
 &nbsp;<font color="#434f54">&#47;&#47; this function can be changed in LCDML_display_menu tab</font>
 &nbsp;<font color="#5e6d03">#define</font> <font color="#00979c">_LCDML_DSIP_use_header</font> <font color="#000000">0</font>

<font color="#95a5a6">&#47;*</font>
<font color="#95a5a6"> &nbsp;void lcdml_menu_display();</font>
<font color="#95a5a6"> &nbsp;void lcdml_menu_clear();</font>
<font color="#95a5a6"> &nbsp;void lcdml_menu_control();</font>
<font color="#95a5a6">*&#47;</font>
<font color="#5e6d03">#include</font> <font color="#005c5f">&#34;menuToDisplay.h&#34;</font>

<font color="#434f54">&#47;&#47;Objekte initialisieren</font>
 &nbsp;<b><font color="#d35400">LCDMenuLib2_menu</font></b> <font color="#d35400">LCDML_0</font> <font color="#000000">(</font><font color="#000000">255</font><font color="#434f54">,</font> <font color="#000000">0</font><font color="#434f54">,</font> <font color="#000000">0</font><font color="#434f54">,</font> <font color="#00979c">NULL</font><font color="#434f54">,</font> <font color="#00979c">NULL</font><font color="#000000">)</font><font color="#000000">;</font> <font color="#434f54">&#47;&#47; root menu element (do not change)</font>
 &nbsp;<b><font color="#d35400">LCDMenuLib2</font></b> <font color="#d35400">LCDML</font><font color="#000000">(</font><font color="#d35400">LCDML_0</font><font color="#434f54">,</font> <font color="#00979c">_LCDML_DISP_rows</font><font color="#434f54">-</font><font color="#00979c">_LCDML_DSIP_use_header</font><font color="#434f54">,</font> <font color="#00979c">_LCDML_DISP_cols</font><font color="#434f54">,</font> <font color="#000000">lcdml_menu_display</font><font color="#434f54">,</font> <font color="#000000">lcdml_menu_clear</font><font color="#434f54">,</font> <font color="#000000">lcdml_menu_control</font><font color="#000000">)</font><font color="#000000">;</font>

</pre>

The headerfile
menuToDisplay.h:

<pre>
<font color="#95a5a6">&#47;* ******************************************************************** *&#47;</font>
<font color="#00979c">void</font> <font color="#000000">lcdml_menu_clear</font><font color="#000000">(</font><font color="#000000">)</font>
<font color="#95a5a6">&#47;* ******************************************************************** *&#47;</font>
<font color="#000000">{</font>
 &nbsp;<font color="#000000">lcd</font><font color="#434f54">.</font><font color="#d35400">clear</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#000000">lcd</font><font color="#434f54">.</font><font color="#d35400">setCursor</font><font color="#000000">(</font><font color="#000000">0</font><font color="#434f54">,</font> <font color="#000000">0</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">}</font>

<font color="#95a5a6">&#47;* ******************************************************************** *&#47;</font>
<font color="#00979c">void</font> <font color="#000000">lcdml_menu_display</font><font color="#000000">(</font><font color="#000000">)</font>
<font color="#95a5a6">&#47;* ******************************************************************** *&#47;</font>
<font color="#000000">{</font>
 &nbsp;<font color="#434f54">&#47;&#47; update content</font>
 &nbsp;<font color="#434f54">&#47;&#47; ***************</font>
 &nbsp;<font color="#5e6d03">if</font> <font color="#000000">(</font><font color="#d35400">LCDML</font><font color="#434f54">.</font><font color="#d35400">DISP_checkMenuUpdate</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">)</font> <font color="#000000">{</font>
 &nbsp;&nbsp;&nbsp;<font color="#434f54">&#47;&#47; clear menu</font>
 &nbsp;&nbsp;&nbsp;<font color="#434f54">&#47;&#47; ***************</font>
 &nbsp;&nbsp;&nbsp;<font color="#d35400">LCDML</font><font color="#434f54">.</font><font color="#d35400">DISP_clear</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>

</pre>
........

So I get always errors, no matter what I declared first. Any ideas?

larryl79, 2022-04-27:
I've got a same issue in vsCode & PlatformIO
framework: Arduino
platform ESP32

Is any solution already?

larryl79, 2022-04-27:
#example code #vscode #platformio #esp32 devkit

I've edited your LCDML_001_liquidCrystal.ino example for platformio (renamed to main.cpp, and included rest of the files as .h after constructor).

src dir:
LCDML_condition.h LCDML_control.h LCDML_display_dynFunction.h LCDML_display_menu.h LCDML_display_menuFunction.h main.cpp

main.cpp (edited part):
`// *********************************************************************
// Objects
// *********************************************************************
LCDMenuLib2_menu LCDML_0 (255, 0, 0, NULL, NULL); // root menu element (do not change)
LCDMenuLib2 LCDML(LCDML_0, _LCDML_DISP_rows-_LCDML_DSIP_use_header, _LCDML_DISP_cols, lcdml_menu_display, lcdml_menu_clear, lcdml_menu_control);

#include <LCDML_control.h>
#include <LCDML_display_menuFunction.h>
#include <LCDML_display_menu.h>
#include <LCDML_condition.h>
#include <LCDML_display_dynFunction.h>

// *********************************************************************
// LCDML MENU/DISP
// *********************************************************************`

(for another try I downloaded the latest lib from github, and overwritten in the libs dir. It's same issue)
when I try compile, I got error messages for the lib itself only.

In file included from C:\Users\larry\.platformio\lib\LCDMenuLib2\src/LCDMenuLib2.h:183:0, from src\main.cpp:24: src/LCDML_display_menu.h: In function 'void lcdml_menu_display()': C:\Users\larry\.platformio\lib\LCDMenuLib2\src/LCDMenuLib2_macros.h:118:29: error: 'g_LCDML_DISP_lang_lcdml_table' was not declared in this scope strcpy(var, g_LCDML_DISP_lang_ ## lang ## _table[id]); \ ^ C:\Users\larry\.platformio\lib\LCDMenuLib2\src/LCDMenuLib2_macros.h:138:9: note: in expansion of macro 'LCDML_getCustomContent' LCDML_getCustomContent(lcdml, var, id) ^ src/LCDML_display_menu.h:50:9: note: in expansion of macro 'LCDML_getContent' LCDML_getContent(content_text, LCDML.MENU_getParentID()); ^ C:\Users\larry\.platformio\lib\LCDMenuLib2\src/LCDMenuLib2_macros.h:118:29: error: 'g_LCDML_DISP_lang_lcdml_table' was not declared in this scope strcpy(var, g_LCDML_DISP_lang_ ## lang ## _table[id]); \ ^ C:\Users\larry\.platformio\lib\LCDMenuLib2\src/LCDMenuLib2_macros.h:138:9: note: in expansion of macro 'LCDML_getCustomContent' LCDML_getCustomContent(lcdml, var, id) ^ src/LCDML_display_menu.h:69:13: note: in expansion of macro 'LCDML_getContent' LCDML_getContent(content_text, tmp->getID()); ^ *** [.pio\build\esp32doit-devkit-v1\src\main.cpp.o] Error 1

larryl79, 2022-04-28:
### **Solution !!!!**

`
extern char\* g_LCDML_DISP_lang_lcdml_table[254];
`

into a right place in the example code:

`// ============================================================
// Example:     LCDML_serialmonitor
// ============================================================

  [...]

//#define _LCDML_cfg_use_ram 

// *********************************************************************
// includes
// *********************************************************************
#include <Arduino.h>
  #include <LCDMenuLib2.h>

**extern char\* g_LCDML_DISP_lang_lcdml_table[254];**

// U8g2lib
  
  #include <U8g2lib.h>

//  #include <Wire.h>
//  #include <LiquidCrystal_I2C.h>
  #ifdef U8X8_HAVE_HW_SPI
  #include <SPI.h>
  #endif
  #ifdef U8X8_HAVE_HW_I2C
  #include <Wire.h>
  #endif 

// *********************************************************************
// U8GLIB
// *********************************************************************
  // U8g2 Constructor List (Frame Buffer)
  // The complete list is available here: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp
  // Please update the pin numbers according to your setup. Use U8X8_PIN_NONE if the reset pin is not connected
  //U8G2_ST7920_128X64_F_HW_SPI u8g2(U8G2_R0, /* CS=*/ 53, /* reset=*/ U8X8_PIN_NONE);     // (MEGA, ...
  //U8G2_ST7920_128X64_F_HW_SPI u8g2(U8G2_R0, /* CS=*/ 12, /* reset=*/ U8X8_PIN_NONE);   // (Uno and co
  U8G2_ST7920_128X64_F_SW_SPI u8g2(U8G2_R0, /* clock=*/ 25, /* data= /R/w */ 33, /* CS=*/ 32, /* reset= */ U8X8_PIN_NONE );  // ESP32

  // settings for u8g lib and LCD
  #define _LCDML_DISP_w                 128            // LCD width
  #define _LCDML_DISP_h                 64             // LCD height
  // font settings
  #define _LCDML_DISP_font              u8g2_font_squeezed_r7_tr   // u8g_font_6x13  // u8glib font (more fonts under u8g.h line 1520 ...)
  #define _LCDML_DISP_font_w            7              // font width
  #define _LCDML_DISP_font_h            8             // font height
  // cursor settings
  #define _LCDML_DISP_cursor_char       ">"            // cursor char
  #define _LCDML_DISP_cur_space_before  2              // cursor space between
  #define _LCDML_DISP_cur_space_behind  4              // cursor space between
  // menu position and size
  #define _LCDML_DISP_box_x0            0              // start point (x0, y0)
  #define _LCDML_DISP_box_y0            0              // start point (x0, y0)
  #define _LCDML_DISP_box_x1            128            // width x  (x0 + width)
  #define _LCDML_DISP_box_y1            64             // hight y  (y0 + height)
  #define _LCDML_DISP_draw_frame        1              // draw a box around the menu
   // scrollbar width
  #define _LCDML_DISP_scrollbar_w       6  // scrollbar width (if this value is < 3, the scrollbar is disabled)

  // nothing change here
  #define _LCDML_DISP_cols_max          ((_LCDML_DISP_box_x1-_LCDML_DISP_box_x0)/_LCDML_DISP_font_w)
  #define _LCDML_DISP_rows_max          ((_LCDML_DISP_box_y1-_LCDML_DISP_box_y0-((_LCDML_DISP_box_y1-_LCDML_DISP_box_y0)/_LCDML_DISP_font_h))/_LCDML_DISP_font_h)

  // rows and cols
  // when you use more rows or cols as allowed change in LCDMenuLib.h the define "_LCDML_DISP_cfg_max_rows" and "_LCDML_DISP_cfg_max_string_length"
  // the program needs more ram with this changes
  #define _LCDML_DISP_rows              _LCDML_DISP_rows_max  // max rows
  #define _LCDML_DISP_cols              20                   // max cols

// *********************************************************************
// Prototypes
// *********************************************************************
  void lcdml_menu_display();
  void lcdml_menu_clear();
  void lcdml_menu_control();

// *********************************************************************
// Objects
// *********************************************************************
  LCDMenuLib2_menu LCDML_0 (255, 0, 0, NULL, NULL); // root menu element (do not change)
  LCDMenuLib2 LCDML(LCDML_0, _LCDML_DISP_rows, _LCDML_DISP_cols, lcdml_menu_display, lcdml_menu_clear, lcdml_menu_control);

  #include <LCDML_display_dynFunction.h>
  #include <LCDML_display_menu.h>
  #include <LCDML_condition.h>
  #include <LCDML_display_menuFunction.h>
  #include <LCDML_control.h>

// *********************************************************************
// LCDML MENU/DISP
// *********************************************************************
  // LCDML_0        => layer 0
  // LCDML_0_X      => layer 1
  // LCDML_0_X_X    => layer 2
  // LCDML_0_X_X_X  => layer 3
  // LCDML_0_...      => layer ...

  // For beginners
  // LCDML_add(id, prev_layer, new_num, lang_char_array, callback_function)

[...]
`

Jomelo, 2022-05-02:
The example code is added to the lib. Thank you for the pull requests.

closed 2022-05-02 by Jomelo