Jomelo/LCDMenuLib2 #38: dynMenu with submenu

Nenadl84, 2019-01-10:
First of all, thank you very much for creating this library.
I have created a dynamical menu item with sub-menu structure that I wish to access by;
` if(LCDML.BT_checkEnter())
  {
    LCDML.FUNC_close(); 
    LCDML.OTHER_setCursorToID(LCDML.FUNC_getID()+1);
    LCDML.BT_resetEnter();                    
  }`
But I can not correctly update the LCD since dyn item seems to be locked. Currently I use this workaround:
` if(LCDML.BT_checkEnter())
  {
    LCDML.FUNC_close(); 
    LCDML.OTHER_jumpToID(LCDML.FUNC_getID()+1);
    LCDML.FUNC_goBackToMenu(0);
    LCDML.BT_resetEnter();                    
  };`
Any advice?

Jomelo, 2019-01-13:
Hello, 
Can you test this version: https://github.com/Jomelo/LCDMenuLib2/tree/develop

I have rewrite the complete library code because there were some bugs which i could not fix on the last code base.

One Line in the LCDML_display_menu tab have been changed.
old:
if ((tmp = LCDML.MENU_getObj()->getChild(LCDML.MENU_getScroll()))
...
new: 
if ((tmp = LCDML.MENU_getDisplayedObj()) != NULL)
...

Nenadl84, 2019-01-14:
Thanks! Now it works with both:
`LCDML.OTHER_setCursorToID();
LCDML.OTHER_setCursorToFunc();`
Just to note, there is an issue with LCDML.FUNC_getID() inside dyn_Func...

Jomelo, 2019-01-14:
I will check this.