Jomelo/LCDMenuLib2 #39: _LCDML_TYPE_dynParam, layer other than LCDML_0

kkaleta, 2019-01-11:
Hi there Jomelo! Really Great library, thank you.

I`m wondering, is the _LCDML_TYPE_dynParam wotks only at 0 layer menu?

I try to define menu like this:

  LCDML_add (8 , LCDML_0_1 , 7  , "External trigger" , NULL);
  LCDML_addAdvanced (9 , LCDML_0_1_1 , 1  , NULL, "" , mDyn_para_steps_togo, 0,_LCDML_TYPE_dynParam); 

..but the function mDyn_para_steps_togo it never starts.

Jomelo, 2019-01-14:
Hello, 

can you initialize the element with the id with this:

LCDML_addAdvanced (9 , LCDML_0_1_7, ...

I have test the dyn elements on a sublayer with the latest beta version. At the moment it seems to run.

kkaleta, 2019-01-15:
I can't compile develop tree version (version=2.0.0 - beta 6)

..\Sketchbook\libraries\LCDMenuLib2-develop\src\LCDMenuLib2.cpp:1332:54: error: no 'uint8_t LCDMenuLib2::MENU_getParentID(uint8_t)' member function declared in class 'LCDMenuLib2'

 uint8_t LCDMenuLib2::MENU_getParentID(uint8_t p_layer)

exit status 1
Error compiling for board Arduino Nano.

Jomelo, 2019-01-15:
Tanks for reporting, its solved. -> beta 7

closed 2019-01-18 by kkaleta