Jomelo/LCDMenuLib2 #40: custum action in specified menu

clear-sky, 2019-01-14:
i use IR remote for my project and i want to add a custom  action in specified menu
 if i add a custom key for my action it works but i can not prevent it when i am in another menu!
is there a way?

Jomelo, 2019-01-14:
Hello, 
you need a backend system. Can you look for the examples with a task system like 01_backend/LCDML_011_TaskSchedular.

Create a small task an start your event when something happen. 

Also you can split the LCDML.loop() function into:

LCDML.loop_control();

// check your special action here 

// call the menu system
LCDML.loop_menu();

Both ways are ok .

clear-sky, 2019-01-15:
hi dear jomelo !
let me to explain more:
i want if i am in setting menu for example key 1 of my irremote must work but when i am in another menu key 1 should not work
thanks for your notice

clear-sky, 2019-01-25:
solved the problem!
for everyone wants how:
1-create a bool value for your condition with 0 value (bool my condition = 0;)
2-set it to 1 in setup section for your menu
3-add an if for your condition in remote section
if (my-condition)
{
do my action
}
4-dont forget to set your bool value to 0  when exit your menu!
thats all

big thanks to jomelo for his great work

closed 2019-01-25 by Jomelo