Jomelo/LCDMenuLib2 #42: Select language during program execution

effeci, 2019-02-02:
Hello! Firstable, Thanks for this beautiful library... I was tinkering with it from version one. I do like how the multilanguage is implemented, but i was wondering if it the language can be changed while using the program, like 
--->settings--->Language---->select Language;
and then going back and ... boom: language switched.
Is it possible? Thanks for the attention.

Jomelo, 2019-02-02:
Yes it is possible with the latest version 2.1.0

In version 2.1.0 is a new example named LCDML_022_multiLanguage and some new macros. When you open this example and changed the variable from 

lang_types g_lcdml_lang_select = LANG_DE;
to 
lang_types g_lcdml_lang_select = LANG_DEFAULT;  

in your program. The language will change directly.

effeci, 2019-02-02:
Sorry but is it possible to switch while the program is running?

Jomelo, 2019-02-02:
yes change the variable in the program.

Jomelo, 2019-02-02:
I add a smaller example to test it:

https://github.com/Jomelo/LCDMenuLib2/tree/master/examples/02_functionality

effeci, 2019-02-02:
Ohhhh now I see! Vielen dank!

closed 2019-02-02 by Jomelo