Jomelo/LCDMenuLib2 #43: Problem with hidden elements and menu update

snoerenberg, 2019-03-13:
Hi,

thanks for your great library.
I use it in a personal project and got the following problem:

- I use a global variable to define a "program" which running (0-4, 0 is off, 1 is 1st program, ...)
- I use a condition to check if a program is running or not and hide/show either the entry to start the program or only the "stop" and "back" function
- A second condition is present to check if any program is running to hide/show the "Stop program" entry

1st problem: Once I started a program I do not get back to the "Stop Program/Back" screen even I Used "LCDML.FUNC_goBackToMenu(1);" which should get me not to the root.

2nd problem: Once I started a program and stopped it again, the reload takes place and I see all 4 programs which I could start, but I can only select between "Program 1" and "Program 2". It seems that the update of the menu does not take care of the hidden elements changed and therefor only the prior present 2 can be selected (prior the screen update only "Stop Program" and "Back" have been present)

I would really like to get there a solution. I could not figure it out, even with a manual work around:

`
// LCDML.OTHER_setCursorToID(2);
// fix for hidden element
// http://forum.arduino.cc/index.php?topic=73816.msg3610953#msg3610953
`

I attach a working sample for the serial monitor here.
[LCDML_000_serialMonitor_Prob_Hidden.zip](https://github.com/Jomelo/LCDMenuLib2/files/2961348/LCDML_000_serialMonitor_Prob_Hidden.zip)

Thanks a lot
Stephan

Jomelo, 2019-03-13:
Hello, 
i can reproduce the problem and try to find a fix. 
Its a bug in the lib.

snoerenberg, 2019-03-19:
Hi @Jomelo is there anything I can help you with? I've not yet compiled the library on my own with debug enabled.

Jomelo, 2019-03-20:
Hello, 
i am working on it, but it is not so easy. 

For example what is correct for the next cursor position
- the buttons programm 1, 2, 3, 4 would be set to hidden, the next and new visible element is "stop program" (seems to be ok)

Example 2:
- between the buttons program 1 and program 2 is a button "stop program". Now i start program 2 and the next visible element did not exists because it is the elment infront of program 2. 

I can program it so, that it worked with your example, but other peaple have the same problem with other menu structures. At the moment the menu goes to the next stable state. A stable state is to set the menu to the root position. 

A possible solution could be to add new function to control the "goBack" handling like this:
- LCDML.FUNC_goBackToNextVisibleElement();
and
- LCDML.FUNC_goBackToFirstVisibleElement();

With this function the user can decide to how the handling is correct. When the next visible element is not found. The menu goes to the layer above and search the next visible element.

At the moment i test this method.

Jomelo, 2019-03-20:
Another solution is to work with  LCDML.OTHER_setCursorToID(...) and set the cursor on the defined position. 

I think it is the easiest way to solve it fastly.

Jomelo, 2019-03-20:
I have edit your example to select the correct cursor position after stopping the programn.

> // *********************************************************************
> void mFunc_startProcess(uint8_t param)
> // *********************************************************************
> {
>   if (LCDML.FUNC_setup()) // ****** SETUP *********
>   {
>      _currentProgram = param;
>      **LCDML.OTHER_setCursorToID(15);**
>     
>     
>     //LCDML.OTHER_setCursorToID(2);
>     // fix for hidden element
>     // http://forum.arduino.cc/index.php?topic=73816.msg3610953#msg3610953
>   }
> 
>   if (LCDML.FUNC_loop()) // ****** LOOP *********
>   {
>     
>   }
> 
>   if (LCDML.FUNC_close()) // ****** STABLE END *********
>   {
>     // you can here reset some global vars or do nothing    
>   }
> }
> 
> // *********************************************************************
> void mFunc_stopProcess(uint8_t param)
> // *********************************************************************
> {
>   if (LCDML.FUNC_setup()) // ****** SETUP *********
>   {
>     **uint8_t cursor_element_id = 3 + ((_currentProgram - 1) * 3);**
>     
>     _currentProgram = 0;
>     
>     LCDML.OTHER_setCursorToID(cursor_element_id);
>   }
> 
>   if (LCDML.FUNC_loop()) // ****** LOOP *********
>   {
>     
>   }
> 
>   if (LCDML.FUNC_close()) // ****** STABLE END *********
>   {
>   }
> }

snoerenberg, 2019-03-23:
Hi @Jomelo, perfect - the workaround works :) also in my CrystalLCD project where I use it

closed 2019-03-23 by Jomelo