Jomelo/LCDMenuLib2 #44: is there a way to have a black page main menu

gcharles81, 2019-04-05:
Hello is there a way to have a black page main menu and only access the main menu with contents on button input thanks

Jomelo, 2019-04-05:
Hello, 
i do not understand your question completly. What is a blank page ? 

Need you a empty screen until a button is pressed ? 
Then create a menu function with no content and jump to this function (on setup). When a button is pressed the function is closed and the main menu is displayed.

gcharles81, 2019-04-12:
thanks for the prompt answer 

Yes I need u an empty screen until a button is pressed ?

Then create a menu function with no content and jump to this function (on setup). When a button is pressed the function is closed and the main menu is displayed.

i am trying that with no success using the LCDML_u8g2lib example

Jomelo, 2019-04-12:
Hey, 

an example: 

File: LCDML_display_menuFunction

Change the mFunc_information to this contet:

```
// *********************************************************************
void mFunc_information(uint8_t param)
// *********************************************************************
{
  if(LCDML.FUNC_setup())          // ****** SETUP *********
  {
    // setup function
    u8g2.setFont(_LCDML_DISP_font);
    u8g2.firstPage();
    do {
      u8g2.drawStr( 0, (_LCDML_DISP_font_h * 1), "dummy content");
      u8g2.drawStr( 0, (_LCDML_DISP_font_h * 2), "empty page");

    } while( u8g2.nextPage() );
  }

  if(LCDML.FUNC_loop())           // ****** LOOP *********
  {    
    // loop function, can be run in a loop when LCDML_DISP_triggerMenu(xx) is set
    // the quit button works in every DISP function without any checks; it starts the loop_end function

    // stay in this function until a button is pressed
    // reset screensaver timer (the screensaver is not called
    LCDML.SCREEN_resetTimer();

    
    if(LCDML.BT_checkAny()) // check if any button is pressed (enter, up, down, left, right)
    {
      // LCDML_goToMenu stops a running menu function and goes to the menu
      LCDML.FUNC_goBackToMenu();
    }
  }

  if(LCDML.FUNC_close())      // ****** STABLE END *********
  {
    // you can here reset some global vars or do nothing
  }
}
```

File: LCDML_u8g2lib

change the void setup() function to this content:

```
// *********************************************************************
// SETUP
// *********************************************************************
  void setup()
  {
    u8g2.begin();

    // serial init; only be needed if serial control is used
    Serial.begin(9600);                // start serial
    Serial.println(F(_LCDML_VERSION)); // only for examples

    // LCDMenuLib Setup
    LCDML_setup(_LCDML_DISP_cnt);

    // Enable Menu Rollover
    LCDML.MENU_enRollover();

    // Enable Screensaver (screensaver menu function, time to activate in ms)
    LCDML.SCREEN_enable(mFunc_screensaver, 10000); // set to 10 seconds
    //LCDML.SCREEN_disable();

    // Some needful methods

    // Jump to the function and stay in it until a button is pressed
    LCDML.OTHER_jumpToFunc(mFunc_information); // the parameter is the function name
  }
```

Now the menu calls the mFunc_information function and stay in it until a button is pressed. The Screensaver is disabled in this function.

gcharles81, 2019-04-12:
Thanks I understand this part now and the LCDML.SCREEN_resetTimer();

Great library by the way and support also

closed 2019-04-12 by Jomelo