Jomelo/LCDMenuLib2 #47: Teensy 3.2 / LCDMenuLib2 / Rotary Encoder Interrupts

medienverbinder, 2019-06-29:
I tried the LCDMenuLib2 [example with an Oled Display and rotary Encoder](https://github.com/Jomelo/LCDMenuLib2/tree/160f3e19d6f203190f030f37a40664db429adf0f/examples/03_displaytypes/gfx/LCDML_adafruit_gfx_ssd1306) on a Teensy 3.2. (Arduino IDE 1.8.9/Tinsyduino 1.46)

Unfortunately the variant on the Teensy only works if I switch off the interrupts ([>> here <<](https://github.com/Jomelo/LCDMenuLib2/blob/160f3e19d6f203190f030f37a40664db429adf0f/examples/03_displaytypes/gfx/LCDML_adafruit_gfx_ssd1306/LCDML_control.ino#L247)):

=> //#define ENCODER_OPTIMIZE_INTERRUPTS
=> #define ENCODER_DO_NOT_USE_INTERRUPTS

(with a small change from Encoder [Issue #40](https://github.com/PaulStoffregen/Encoder/issues/40), the example on Node MCU (ESP 8266) works.)

I'm not sure where the error is, so there is also a corresponding issue on the Encoder GitHub page:

=> https://github.com/PaulStoffregen/Encoder/issues/41

Does anyone know a workaround so that the interrupts can be used or does anyone have a suggestion why the navigation doesn't work anymore when the interrupts are activated?

Zefram88, 2019-07-30:
I've solved comment out those two lines:
    //pinMode(encoder_A_pin      , INPUT_PULLUP);
    //pinMode(encoder_B_pin      , INPUT_PULLUP);

Now encoder with interrupts works

medienverbinder, 2019-08-09:
@Zefram88 Thanks for the tip, now everything works.
 I have adjusted the following settings:

`
#define ENCODER_USE_INTERRUPTS
#define ENCODER_OPTIMIZE_INTERRUPTS //Only when using pin2/3 (yes, I use them. :-) )

//pinMode(encoder_A_pin      , INPUT_PULLUP);
//pinMode(encoder_B_pin      , INPUT_PULLUP);
`

closed 2019-08-09 by medienverbinder