Jomelo/LCDMenuLib2 #49: LCDMenuLib2 + TFT ST7735

framal15, 2019-07-14:
Hallo Jomelo,
ich benutze deine MenuLib schon länger mit LCD-Displays. Echt super! Jetzt möchte ich auf TFT-isplays mit einen ST7735 Controller umsteigen. Ich nutze 1.8 und 1.77" Displays (128x160).
Im Prinzip funktioniert das auch. Nur der Cursor(X) überschreibt jedes mal den 1. Buchstaben der Menüeinträge und ich sehe keinen Scrollbalken?! Muss ich noch Parameter ändern?

Jomelo, 2019-07-14:
Moin, 
ich habe mir den Beispiel code gerade angesehen und festgestellt das ich den Scrollbarcode bewusst noch nicht aktiviert habe 

 #ifdef _SCROLLBAR_TODO_
...

Dort ist noch ein Bug den ich beheben sollte. 
Im Moment habe ich keinen Arduino und auch kein Display lauffähig hier. Ich müsste mir erst einen Aufbau zusammenstellen um diesem Part einzubinden. 

Viele Grüße

framal15, 2019-07-15:
Moin,
vielen Dank für die Antwort. Gibt es auch schon eine Idee warum der Cursor immer den 1. Buschstaben überschreibt?

Jomelo, 2019-07-15:
Moin, 
ich habe den Code noch nicht angepasst. Im Moment habe ich kein Display hier, daher kann ich es auch nicht testen oder simulieren.

framal15, 2019-07-15:
Ok. Das mit dem Cursor hab ich schon selbst gefunden:

für reguläre Menueinträge in "LCDML_display_menu.ino":
do
    {
      // check if a menu element has a condition and if the condition be true
      if (tmp->checkCondition())
      {
        // check the type off a menu element
        if(tmp->checkType_menu() == true)
        {
          // display normal content
          LCDML_getContent(content_text, tmp->getID());
          display.setCursor(10, _LCDML_ADAFRUIT_FONT_H * (n));               -----> "10"  statt "0" (Pixel)

für dy. Einträge in "LDML_display_dynFunction.ino":

char buf[20];
  sprintf (buf, "dynValue: %d", g_dynParam);

  display.setCursor(10, _LCDML_ADAFRUIT_FONT_H * (line));   -----------> das gleiche!
  display.println(buf);
}