Jomelo/LCDMenuLib2 #52: SAMD support

JustSomeRandomDIYGuy, 2020-01-24:
Hi Jomelo, 

I've used your lib now for different projects and the most are running on SAMD21 boards like Adafruit's Feather M0. Yesterday I tried the new Feather M4 Express with the SAMD51 and it also compiles and runs without any issue. 

The only thing that disturbs a bit are the compiler warnings during compilation in Arduino IDE and the compiler aborts in Visual Micro. I can fix both by modifying your library.properties file but I guess its not necessary to exclude this architecture.

Do you think about lifting this limitation or is there some hard reason for not supporting those chips that I am not aware of?

Regards,
Marco

Jomelo, 2020-01-24:
Hello, 
i have changed the architecture flag to "*". There is no limitation. The new version is available on master branch. 
Regards,
Nils

JustSomeRandomDIYGuy, 2020-01-25:
Great. Thank you!

closed 2020-01-25 by JustSomeRandomDIYGuy