Jomelo/LCDMenuLib2 #55: IRMP support

clear-sky, 2020-02-13:
hi dear jomelo
IRremote has no update for several years but irmp is newer than irremote and is more more powerful than it 
have you any plan to replace irremote with irmp?
https://github.com/ukw100/IRMP

clear-sky, 2020-04-13:
this is my change in "" LCDML_control " and irmp works fine
#elif(_LCDML_CONTROL_cfg == 5)
#define IRMP_INPUT_PIN PA0
#define IRMP_PROTOCOL_NAMES              1 // Enable protocol number mapping to protocol strings - needs some FLASH. Must before #include <irmp*>
#include <irmpSelectMain15Protocols.h>  // This enables 15 main protocols

#include <irmp.c.h>

IRMP_DATA irmp_data[1];

#define STR_HELPER(x) #x
#define STR(x) STR_HELPER(x)

void handleReceivedIRData();

// *********************************************************************
// change in this function the IR values to your values

void lcdml_menu_control(void)
{
  // If something must init, put in in the setup condition
  if (LCDML.BT_setup()) {
    // runs only once
    //irrecv.enableIRIn(); // Start the receiver
    Serial.begin(115200);
    irmp_init();
     

    Serial.println(F("Ready to receive IR signals at pin " STR(IRMP_INPUT_PIN)));
  }

  if (irmp_get_data(&irmp_data[0]))
  {
    // comment this line out, to check the correct code
    //Serial.println(results.value, HEX);

    // in this switch case you have to change the value 0x...1 to the correct IR code
    
    switch (irmp_data[0].command)
    {
      case 0x52: LCDML.BT_enter(); break;
      case 0x50: LCDML.BT_up();    break;
      case 0x51: LCDML.BT_down();  break;
      case 0x55: LCDML.BT_left();  break;
      case 0x56: LCDML.BT_right(); break;
      case 0x23: LCDML.BT_quit();  break;
      default: break;
    }
    //irrecv.resume(); // Receive the next value
    
  }
}

Jomelo, 2020-04-20:
Hi, 
i have update the examples for the new versionn v2.2.0  and changed the ir control part.

closed 2020-04-20 by Jomelo