Jomelo/LCDMenuLib2 #56: stm32 menu item corruption

clear-sky, 2020-03-20:
hi dear jomelo
i have a arm base board with stm32f103ret6 mcu
and use a color lcd screen ili9486 (480x320) with adafruit gfx driver
i tried this lib and works but have strange problem!
i created a menu item and called a function with it
when i use a simple command for example draw a rectangle everything work well
but if i use several commands i can see that the characters in menu item has Corruption like https://github.com/Jomelo/LCDMenuLib2/issues/54
even before calling that item!!!
the more commands in my function cause more corruption in menu items 
and more of that cause disappearing menu items
i have used your excellent lib in avr  and have not seen this problem atall
i changed #define _LCDML_cfg_use_ram to 1 but no success
please help

clear-sky, 2020-03-27:
installed LCDMenuLib2 ver 2.1.2 from library manager and problem solved
just have a warning as below

WARNING: library LCDMenuLib2 claims to run on avr, esp8266, esp32 architecture(s) and may be incompatible with your current board which runs on stm32 architecture(s).
but problem solved 
in latest git version in lcdmenulib2.h line 95
 #define _LCDML_USE_RAM  refers to nothing
but in stable version 
#define _LCDML_ESP
please close this issue if fixing that line

Jomelo, 2020-04-04:
Hello, 
thank you for your time to search this bug. I will fix it on the next version. 

Regards

Jomelo, 2020-04-20:
Hi, 
i have changed this part. 

All examples have now this comment above the include part of the library

```
// *********************************************************************
// special settings for non arduino basic controllers
// *********************************************************************
// enable this line when you are not usigng a standard arduino
// for example when your chip is an ESP or a STM or SAM or something else

//#define _LCDML_cfg_use_ram 

// *********************************************************************
// includes
// *********************************************************************
  #include <LCDMenuLib2.h>
```

There are so many different defines for all controller architectures that i will not add all to my library.

closed 2020-04-20 by Jomelo