Jomelo/LCDMenuLib2 #57: error

totza2010, 2020-04-18:
LCDML_add         (0  , LCDML_0         , 1  , "Module"      , NULL);
LCDML_add         (1  , LCDML_0_1         , 1  , "Data Logging"        , NULL);
LCDML_add         (2  , LCDML_0_1_1         , 1  , "Configuration Item"        , NULL);
LCDML_add         (3  , LCDML_0_1_1_1         , 1  , "Logging Data"        , NULL);
LCDML_add         (4  , LCDML_0_1_1         , 2  , "Options"        , NULL);
LCDML_add         (5  , LCDML_0_1_1_2         , 1  , "Log Interval"        , NULL);
LCDML_add         (6  , LCDML_0_1         , 2  , "Date and Time"          , NULL);
LCDML_add         (7  , LCDML_0_1_2         , 1  , "Set Date and Time"          , NULL);
LCDML_add         (8  , LCDML_0_1_2         , 2  , "Set to NTP Time"          , NULL);
LCDML_add         (9  , LCDML_0         , 2  , "Device"        , NULL);
LCDML_add         (10 , LCDML_0_2         , 1  , "Options"        , NULL);
LCDML_add         (11 , LCDML_0_2         , 2  , "Calibration"        , NULL);
LCDML_add         (12 , LCDML_0_2_2         , 1  , "Temperature"        , NULL);
LCDML_add         (13 , LCDML_0_2_2_1         , 1  , "Offset"        , NULL);
LCDML_add         (14 , LCDML_0_2_2         , 2  , "Humidity"        , NULL);
LCDML_add         (15 , LCDML_0_2_2_2         , 1  , "Offset"        , NULL);
LCDML_add         (16 , LCDML_0         , 3  , "Communications"         , NULL);
LCDML_add         (17 , LCDML_0_3         , 1  , "WiFi"      , NULL);
LCDML_add         (18 , LCDML_0_3_1         , 1  , "Options"        , NULL);
LCDML_add         (19 , LCDML_0_3_1_1         , 1  , "WiFi Mode"        , NULL);
LCDML_add         (20 , LCDML_0_3_1         , 2  , "Configuration"        , NULL);
LCDML_add         (21 , LCDML_0_3_1_2         , 1  , "Access Point"        , NULL);
LCDML_add         (22 , LCDML_0_3_1_2_1         , 1  , "SSID"        , NULL);
LCDML_add         (23 , LCDML_0_3_1_2_1         , 2  , "Password"        , NULL);
LCDML_add         (24 , LCDML_0_3_1_2_1         , 3  , "Advance"        , NULL);
LCDML_add         (25 , LCDML_0_3_1_2_1_3         , 1  , "Local IP"        , NULL);
LCDML_add         (26 , LCDML_0_3_1_2_1_3         , 2  , "Gateway"        , NULL);
LCDML_add         (27 , LCDML_0_3_1_2_1_3         , 3  , "Subnet"        , NULL);
LCDML_add         (28 , LCDML_0_3_1_2_1_3         , 4  , "Chanel"        , NULL);
LCDML_add         (29 , LCDML_0_3_1_2_1_3         , 5  , "Hidden"        , NULL);
LCDML_add         (30 , LCDML_0_3_1_2         , 2  , "Station"        , NULL);
LCDML_add         (31 , LCDML_0_3_1_2_2         , 1  , "SSID"        , NULL);
LCDML_add         (32 , LCDML_0_3_1_2_2         , 2  , "Password"        , NULL);
LCDML_add         (33 , LCDML_0_3_1_2_2         , 3  , "Advance"        , NULL);
LCDML_add         (34 , LCDML_0_3_1_2_2_3         , 1  , "Local IP"        , NULL);
LCDML_add         (35 , LCDML_0_3_1_2_2_3         , 2  , "Gateway"        , NULL);
LCDML_add         (36 , LCDML_0_3_1_2_2_3         , 3  , "Subnet"        , NULL);
LCDML_add         (37 , LCDML_0_3_1_2_2_3         , 4  , "DNS"        , NULL);
LCDML_add         (38 , LCDML_0_3         , 2  , "RS485"             , NULL);
LCDML_add         (39 , LCDML_0_3_2         , 1  , "Options"        , NULL);
LCDML_add         (40 , LCDML_0_3_2_1         , 1  , "Slave ID"        , NULL);
LCDML_add         (41 , LCDML_0_3_2_1         , 2  , "Baud Rate"        , NULL);
LCDML_addAdvanced (42 , LCDML_0         , 4  , COND_hide,         "screensaver"    , mFunc_screensaver,   0,  _LCDML_TYPE_default);
#define _LCDML_DISP_cnt    42

When set up as this, then hang on the page menu all the time, refusing to go to the page screensever

Jomelo, 2020-04-18:
Hello, 
thanks you for reporting this bug. I have fixed the bug in the last beta version. 

The Problem was the "back jump" over more then one layer. For example at this position:

LCDML_add (5 , LCDML_0_1_1_2 , 1 , "Log Interval" , NULL);
LCDML_add (6 , LCDML_0_1 , 2 , "Date and Time" , NULL);

or on this position:
 
LCDML_add (15 , LCDML_0_2_2_2 , 1 , "Offset" , NULL);
LCDML_add (16 , LCDML_0 , 3 , "Communications" , NULL);

I think the problem is now fixed. Please download the last version from master branch.

totza2010, 2020-04-18:
When update last version

In file included from C:\Users\chalo\OneDrive\COM\Documents\Arduino\libraries\LCDMenuLib2-master\src/LCDMenuLib2.h:191:0,

                 from C:\Users\chalo\OneDrive\COM\Documents\Arduino\LCDML_u8g2lib\LCDML_u8g2lib.ino:4:

C:\Users\chalo\OneDrive\COM\Documents\Arduino\LCDML_u8g2lib\LCDML_display_menu.ino: In function 'void lcdml_menu_display()':

C:\Users\chalo\OneDrive\COM\Documents\Arduino\libraries\LCDMenuLib2-master\src/LCDMenuLib2_macros.h:84:29: error: 'g_LCDML_DISP_lang_lcdml_table' was not declared in this scope

                 strcpy(var, g_LCDML_DISP_lang_ ## lang ## _table[id]); \

                             ^

C:\Users\chalo\OneDrive\COM\Documents\Arduino\libraries\LCDMenuLib2-master\src/LCDMenuLib2_macros.h:104:9: note: in expansion of macro 'LCDML_getCustomContent'

         LCDML_getCustomContent(lcdml, var, id)

         ^

C:\Users\chalo\OneDrive\COM\Documents\Arduino\LCDML_u8g2lib\LCDML_display_menu.ino:30:13: note: in expansion of macro 'LCDML_getContent'

             LCDML_getContent(content_text, tmp->getID());

             ^

Jomelo, 2020-04-19:
Hello, 
in your library path, are there two lcdmenulib2 . The orginal one and the beta ? 

Please save the orginal version (not rename) and remove it. Then rename LCDMenuLib2-master to LCDMenuLib2.

I do not know your versions number of your implementation. I think that was an older one and the arduino IDE try to use the first header file which is found.  From Version 2.0.x to version 2.1.x the multi language support was added. From 2.1.x to 2.2.0 beta there are no changes on this part. 

At the moment i can not reproduce this error message. I have only installed the beta version and test it with arduino and esp with u8g2lib.

totza2010, 2020-04-19:
// *********************************************************************
// LCDML MENU/DISP
// *********************************************************************
LCDML_addAdvanced (0  , LCDML_0         , 1,      NULL, "Module"            ,              NULL, 0, _LCDML_TYPE_default);
LCDML_addAdvanced (1  , LCDML_0_1       , 1,      NULL,   "Data Logging"      ,              NULL, 0, _LCDML_TYPE_default);
LCDML_addAdvanced (2  , LCDML_0_1_1     , 1,      NULL,     "Configuration Item",              NULL, 0, _LCDML_TYPE_default);
LCDML_addAdvanced (3  , LCDML_0_1_1_1   , 1,      NULL,       "Logging Data",              NULL, 0, _LCDML_TYPE_default);
LCDML_addAdvanced (4  , LCDML_0_1_1     , 2,      NULL,     "Options",              NULL, 0, _LCDML_TYPE_default);
LCDML_addAdvanced (5  , LCDML_0_1_1_2   , 1,      NULL,       "Log Interval",              NULL, 0, _LCDML_TYPE_default);
LCDML_addAdvanced (6  , LCDML_0_1       , 2,      NULL,   "Date and Time"     ,              NULL, 0, _LCDML_TYPE_default);
LCDML_addAdvanced (7  , LCDML_0_1_2     , 1,      NULL,     "Set Date and Time" ,              NULL, 0, _LCDML_TYPE_default);
LCDML_addAdvanced (8  , LCDML_0_1_2     , 2,      NULL,     "Set to NTP Time"   ,              NULL, 0, _LCDML_TYPE_default);
LCDML_addAdvanced (9  , LCDML_0         , 2,      NULL, "Device"            ,              NULL, 0, _LCDML_TYPE_default);
LCDML_addAdvanced (10 , LCDML_0         , 3,      NULL, "Communications"    ,              NULL, 0, _LCDML_TYPE_default);
LCDML_addAdvanced (11 , LCDML_0         , 4, COND_hide, "screensaver"       , mFunc_screensaver, 0, _LCDML_TYPE_default);
#define _LCDML_DISP_cnt    11
LCDML_createMenu(_LCDML_DISP_cnt);

When set up as this, not show scrollbar

Jomelo, 2020-04-20:
Hello
when you are looking in the tab / in the file  "LCDML_display_menu" you can manualy change this part. 

`
...
// display scrollbar when more content as rows available with _LCDML_DISP_scrollbar_w  > 2
    if (scrollbar_max > n_max && _LCDML_DISP_scrollbar_w > 2)
    {
...
`

I do not know how many rows are exists in your setup

This is not a bug.

Jomelo, 2020-04-20:
When you found a new bug please report it. I will close this issue.

closed 2020-04-20 by Jomelo