Jomelo/LCDMenuLib2 #59: Support for C++ callbacks

moritz89, 2020-04-20:
Has the possibility been investigated to add support for C++ style callbacks? std::bind from <functional>?

This would allow the callbacks to be realized as member classes, thereby being able to use member variables with references to the LCDMenuLib2 and U8G2.

Jomelo, 2020-04-20:
Hello, 
i am comming from the c world, i think i have to read some theory parts to work with this things ;-)

moritz89, 2020-04-20:
Oh, then it is a quite big change. Fundamentally, it is not possible to cast a C++ function to a C style function. The closest workaround that I've found is creating a C function wrapper to call my C++ function:

    void mFunc_information(uint8_t param) { lcdml_utils.information(param); }
    LCDML_add(0, LCDML_0, 1, "Information", mFunc_information);

`mFunc_information` is passed to `LCDML_add` and `lcdml_utils` is an instance of the `LcdmlUtils` class that I wrote.

The reason I'm asking is because the TaskScheduler library also has a object oriented (C++) interface.

closed 2020-11-30 by Jomelo