Jomelo/LCDMenuLib2 #60: freeze when jump to func

clear-sky, 2020-04-20:
hi dear jomelo
entire program freezes when call LCDML.OTHER_jumpToFunc
(using stm32 in ram mode)

Jomelo, 2020-04-21:
Hi, 
can you enable the debug mode in LCDMenuLib2.h  line 50 and sent me the output which is displayed in the serial monitor ?

```
    // enable debug strings (remove comments from this line)
    #define LCDML_DBG                          1
```
And enable the following debug outputs:
```
    // debug special method groups          // enable a flag to control the function call order
    #define LCDML_DBG_function_name_LOOP        0
    #define LCDML_DBG_function_name_MENU        0
    #define LCDML_DBG_function_name_FUNC        0
    #define LCDML_DBG_function_name_BT          0
    #define LCDML_DBG_function_name_CE          0
    #define LCDML_DBG_function_name_OTHER       1
    #define LCDML_DBG_function_name_DISP        0
    #define LCDML_DBG_function_name_TIMER       0
    #define LCDML_DBG_function_name_SCREEN      0
```

clear-sky, 2020-04-21:
hi dear jomelo and thanks for your notice
here is my console output 
21:34:46.389 -> LCDML2 v2.2.1
21:34:47.217 -> CC⸮⸮Nhj⸮s-]⸮⸮EK`g3⸮cnMl}⸮
[image]

clear-sky, 2020-04-22:
for more info:
after freeze
exit key does not work 
enter key run my app again
left - right -up - down work like exit key (only on my frozen app)

Jomelo, 2020-04-23:
Hello, 
i cannot found the error with this information.

Work the examples correctly ? 

Can you give me:
- your current menu structure (the definition of your menu elements)
- the information if the controler freeze / restarts or only the running program
- does the examples work correctly ?
- are there any compiler warnings ?

clear-sky, 2020-04-24:
your current menu structure (the definition of your menu elements)
LCDML_add         (0    , LCDML_0           , 1  , "CLOCK"            , mFunc_Show_Clock);
LCDML_add         (1    , LCDML_0           , 2  , "SETTINGS"         , NULL);
LCDML_add         (2    , LCDML_0_2         , 1  , "OSCILATOR"        , mFunc_Set_Oscilator);
LCDML_add         (3    , LCDML_0_2         , 2  , "TIME"             , mFunc_Set_Time);
LCDML_add         (4    , LCDML_0_2         , 3  , "CALENDAR"         , mFunc_Set_Calendar);
LCDML_add         (5    , LCDML_0_2         , 4  , "ALARM"            , NULL);
LCDML_add         (6    , LCDML_0_2_4       , 1  , "ALARM1"           , NULL);
LCDML_add         (7    , LCDML_0_2_4_1     , 1  , "ALARM1 TIME"      , mFunc_Alarm1_Time);
LCDML_add         (8    , LCDML_0_2_4_1     , 2  , "ALARM1 SOUND"     , NULL);
LCDML_add         (9    , LCDML_0_2_4_1     , 3  , "ALARM1 RHYTHM"    , mFunc_Alarm1_Rhythm);
LCDML_add         (10   , LCDML_0_2_4       , 2  , "ALARM2"           , NULL);
LCDML_add         (11   , LCDML_0_2_4_2     , 1  , "ALARM2 TIME"      , mFunc_Alarm2_Time);
LCDML_add         (12   , LCDML_0_2_4_2     , 2  , "ALARM2 SOUND"     , NULL);
LCDML_add         (13   , LCDML_0_2_4_2     , 3  , "ALARM2 RHYTHM"    , mFunc_Alarm2_Rhythm);
LCDML_add         (14   , LCDML_0_2_4       , 3  , "LOCAL BEEPS"      , NULL);
LCDML_add         (15   , LCDML_0_2_4       , 4  , "ALARM SCREEN"     , mFunc_Show_Alarm_Screen);
#define _LCDML_DISP_cnt    15

the information if the controler freeze / restarts or only the running program
only the running program

does the examples work correctly ?
yes

are there any compiler warnings ?
there are always compiler warnings
i put all my compiler warnings both ver2.1.2 and 2.2.1 of your lib

i have made a clock with color lcd with this controler :ili9486 and RTC:ds3234 
my platform is stm32(stm32f103ret6)

my jump to func is used to jump from clock menu to alarm menu when alarm fires
in ver 2.1.2 jump to func works but in 2.2.1 clock freezes when alarm is activated
[2.1.2.zip](https://github.com/Jomelo/LCDMenuLib2/files/4527105/2.1.2.zip)
[2.2.1.zip](https://github.com/Jomelo/LCDMenuLib2/files/4527109/2.2.1.zip)

Jomelo, 2020-05-04:
Hi, 
i cannot reproduce this problem at the moment. I have try to remove some deadlocks in version 2.2.2 can you test this version ?

Can you enable more debug informations (not all)

    #define LCDML_DBG_function_name_LOOP        0
    #define LCDML_DBG_function_name_MENU        1
    #define LCDML_DBG_function_name_FUNC        1
    #define LCDML_DBG_function_name_BT          0
    #define LCDML_DBG_function_name_CE          0
    #define LCDML_DBG_function_name_OTHER       1
    #define LCDML_DBG_function_name_DISP        0
    #define LCDML_DBG_function_name_TIMER       0
    #define LCDML_DBG_function_name_SCREEN      0

clear-sky, 2020-05-04:
hi dear jomelo and thanks for your notice
this is output of debug screen from about 2 seconds to freeze 
02:26:58.028 -> LCDML.FUNC_loop
02:26:58.028 -> LCDML.FUNC_close
02:26:58.095 -> LCDML.FUNC_call
02:26:58.095 -> LCDML.FUNC_setup
02:26:58.128 -> LCDML.FUNC_loop
02:26:58.128 -> LCDML.FUNC_close
02:26:58.194 -> LCDML.FUNC_call
02:26:58.194 -> LCDML.FUNC_setup
02:26:58.227 -> LCDML.FUNC_loop
02:26:58.227 -> LCDML.FUNC_close
02:26:58.294 -> LCDML.FUNC_call
02:26:58.294 -> LCDML.FUNC_setup
02:26:58.327 -> LCDML.FUNC_loop
02:26:58.327 -> LCDML.FUNC_close
02:26:58.393 -> LCDML.FUNC_call
02:26:58.393 -> LCDML.FUNC_setup
02:26:58.426 -> LCDML.FUNC_loop
02:26:58.426 -> LCDML.FUNC_close
02:26:58.493 -> LCDML.FUNC_call
02:26:58.493 -> LCDML.FUNC_setup
02:26:58.526 -> LCDML.FUNC_loop
02:26:58.526 -> LCDML.FUNC_close
02:26:58.592 -> LCDML.FUNC_call
02:26:58.592 -> LCDML.FUNC_setup
02:26:58.625 -> LCDML.FUNC_loop
02:26:58.625 -> LCDML.FUNC_close
02:26:58.692 -> LCDML.FUNC_call
02:26:58.692 -> LCDML.FUNC_setup
02:26:58.725 -> LCDML.FUNC_loop
02:26:58.725 -> LCDML.FUNC_close
02:26:58.791 -> LCDML.FUNC_call
02:26:58.791 -> LCDML.FUNC_setup
02:26:58.825 -> LCDML.FUNC_loop
02:26:58.825 -> LCDML.FUNC_close
02:26:58.891 -> LCDML.FUNC_call
02:26:58.891 -> LCDML.FUNC_setup
02:26:58.924 -> LCDML.FUNC_loop
02:26:58.924 -> LCDML.FUNC_close
02:26:58.990 -> LCDML.FUNC_call
02:26:58.990 -> LCDML.FUNC_setup
02:26:59.024 -> LCDML.FUNC_loop
02:26:59.024 -> LCDML.FUNC_close
02:26:59.090 -> LCDML.FUNC_call
02:26:59.090 -> LCDML.FUNC_setup
02:26:59.123 -> LCDML.FUNC_loop
02:26:59.123 -> LCDML.FUNC_close
02:26:59.190 -> LCDML.FUNC_call
02:26:59.190 -> LCDML.FUNC_setup
02:26:59.223 -> LCDML.FUNC_loop
02:26:59.223 -> LCDML.FUNC_close
02:26:59.289 -> LCDML.FUNC_call
02:26:59.289 -> LCDML.FUNC_setup
02:26:59.322 -> LCDML.FUNC_loop
02:26:59.322 -> LCDML.FUNC_close
02:26:59.389 -> LCDML.FUNC_call
02:26:59.389 -> LCDML.FUNC_setup
02:26:59.422 -> LCDML.FUNC_loop
02:26:59.422 -> LCDML.FUNC_close
02:26:59.488 -> LCDML.FUNC_call
02:26:59.488 -> LCDML.FUNC_setup
02:26:59.521 -> LCDML.FUNC_loop
02:26:59.521 -> LCDML.FUNC_close
02:26:59.588 -> LCDML.FUNC_call
02:26:59.588 -> LCDML.FUNC_setup
02:26:59.621 -> LCDML.FUNC_loop
02:26:59.621 -> LCDML.FUNC_close
02:26:59.687 -> LCDML.FUNC_call
02:26:59.687 -> LCDML.FUNC_setup
02:26:59.720 -> LCDML.FUNC_loop
02:26:59.720 -> LCDML.FUNC_close
02:26:59.787 -> LCDML.FUNC_call
02:26:59.787 -> LCDML.FUNC_setup
02:26:59.820 -> LCDML.FUNC_loop
02:26:59.820 -> LCDML.FUNC_close
02:26:59.886 -> LCDML.FUNC_call
02:26:59.886 -> LCDML.FUNC_setup
02:26:59.953 -> LCDML.FUNC_loop
02:26:59.953 -> LCDML.FUNC_close
02:26:59.986 -> LCDML.FUNC_call
02:26:59.986 -> LCDML.FUNC_setup
02:27:00.052 -> LCDML.FUNC_loop
02:27:00.052 -> LCDML.OTHER_jumpToFunc
02:27:00.085 -> LCDML.OTHER_jumpToFunc - start
02:27:00.185 -> LCDML.FUNC_close
02:27:00.185 -> LCDML.OTHER_jumpTo.. / LCDML.OTHER_setCursorTo... - is active
02:27:00.251 -> LCDML.MENU_enScroll
02:27:00.317 -> LCDML.OTHER_jumpTo.. / LCDML.OTHER_setCursorTo... - close active function
02:27:00.384 -> LCDML.FUNC_call
02:27:00.417 -> LCDML.FUNC_setup
02:27:00.417 -> LCDML.FUNC_loop
02:27:00.450 -> LCDML.FUNC_close
02:27:00.450 -> LCDML.MENU_resetActiveMenu
02:27:00.483 -> LCDML.MENU_getRootObj
02:27:00.516 -> ID: 0
02:27:00.516 -> check function
02:27:00.550 -> ID: 1
02:27:00.550 -> check function
02:27:00.550 -> LCDML.OTHER_jumpTo.. / LCDML.OTHER_setCursorTo... - nothing found

Jomelo, 2020-05-05:
Hi, 
on the newest master version this problem is fixed:   v2.2.3

clear-sky, 2020-05-11:
hi dear jomelo
thanks for your notice
i confirm that problem is solved
you are great

closed 2020-05-11 by clear-sky