Jomelo/LCDMenuLib2 #62: Trying to control with buttons and joystick

romton843, 2020-04-29:
Hi, first thing, I wanted to sincerely thank you for your amazing job on this awesome library.
I tried to build my own set of controls to be able to use 6 buttons and a joystick in same time (because in some parts of my menu, a joystick is better suited and in some other parts, buttons are and I needed a joystick anyway). However, with my new set of _LCDML_CONTROL_cfg which almost works, there are moments where using the joystick makes everything freeze.
Did someone tried that and encounter the same problem? Is it not recommended to do so and should I just go with a button config and creating events for the joystick? Will a combination of several controls be an option some day, like buttons and IR, keypad and joystick...?
Best regards. Romain

romton843, 2020-04-29:
Actually, it seems to work now, I might have done it slightly differently on this attempt.
Let me know if you think of any problem that might occur.
Otherwise, the case is closed I guess.
Best regards.
Romain 

// *********************************************************************
// ******************** (8) CUSTOM CONTROL *****************************
// *********************************************************************
#elif(_LCDML_CONTROL_cfg == 8)
// settings
  unsigned long g_LCDML_DISP_press_time = 0;

  #define _LCDML_CONTROL_digital_low_active      1    // 0 = high active (pulldown) button, 1 = low active (pullup)
                                                      // http://playground.arduino.cc/CommonTopics/PullUpDownResistor
  #define _LCDML_CONTROL_digital_enable_quit     1
  #define _LCDML_CONTROL_digital_enable_lr       1
  #define _LCDML_CONTROL_digital_enter           7
  #define _LCDML_CONTROL_digital_up              5
  #define _LCDML_CONTROL_digital_down            4
  #define _LCDML_CONTROL_digital_quit            6
  #define _LCDML_CONTROL_digital_left            2
  #define _LCDML_CONTROL_digital_right           3
    
  #define _LCDML_CONTROL_analog_pinx A0
  #define _LCDML_CONTROL_analog_piny A1

  // when you did not use a button set the value to zero
  #define _LCDML_CONTROL_analog_up_min 612 // Button Up
  #define _LCDML_CONTROL_analog_up_max 1023
  #define _LCDML_CONTROL_analog_down_min 0 // Button Down
  #define _LCDML_CONTROL_analog_down_max 412
  #define _LCDML_CONTROL_analog_left_min 0 // Button Left
  #define _LCDML_CONTROL_analog_left_max 412
  #define _LCDML_CONTROL_analog_right_min 612 // Button Right
  #define _LCDML_CONTROL_analog_right_max 1023
// *********************************************************************
void lcdml_menu_control(void)
{
  // If something must init, put in in the setup condition
  if(LCDML.BT_setup()) {
    // runs only once
    // init buttons
    pinMode(_LCDML_CONTROL_digital_enter      , INPUT_PULLUP);
    pinMode(_LCDML_CONTROL_digital_up         , INPUT_PULLUP);
    pinMode(_LCDML_CONTROL_digital_down       , INPUT_PULLUP);
    # if(_LCDML_CONTROL_digital_enable_quit == 1)
      pinMode(_LCDML_CONTROL_digital_quit     , INPUT_PULLUP);
    # endif
    # if(_LCDML_CONTROL_digital_enable_lr == 1)
      pinMode(_LCDML_CONTROL_digital_left     , INPUT_PULLUP);
      pinMode(_LCDML_CONTROL_digital_right    , INPUT_PULLUP);
    # endif
  }

  #if(_LCDML_CONTROL_digital_low_active == 1)
  #  define _LCDML_CONTROL_digital_a !
  #else
  #  define _LCDML_CONTROL_digital_a
  #endif

  uint8_t but_stat = 0x00;

  bitWrite(but_stat, 0, _LCDML_CONTROL_digital_a(digitalRead(_LCDML_CONTROL_digital_enter)));
  bitWrite(but_stat, 1, _LCDML_CONTROL_digital_a(digitalRead(_LCDML_CONTROL_digital_up)));
  bitWrite(but_stat, 2, _LCDML_CONTROL_digital_a(digitalRead(_LCDML_CONTROL_digital_down)));
  #if(_LCDML_CONTROL_digital_enable_quit == 1)
  bitWrite(but_stat, 3, _LCDML_CONTROL_digital_a(digitalRead(_LCDML_CONTROL_digital_quit)));
  #endif
  #if(_LCDML_CONTROL_digital_enable_lr == 1)
  bitWrite(but_stat, 4, _LCDML_CONTROL_digital_a(digitalRead(_LCDML_CONTROL_digital_left)));
  bitWrite(but_stat, 5, _LCDML_CONTROL_digital_a(digitalRead(_LCDML_CONTROL_digital_right)));
  #endif

  if (but_stat > 0) {
    if((millis() - g_LCDML_DISP_press_time) >= 200) {
      g_LCDML_DISP_press_time = millis(); // reset press time

      if (bitRead(but_stat, 0)) { LCDML.BT_enter(); }
      if (bitRead(but_stat, 1)) { LCDML.BT_up();    }
      if (bitRead(but_stat, 2)) { LCDML.BT_down();  }
      if (bitRead(but_stat, 3)) { LCDML.BT_quit();  }
      if (bitRead(but_stat, 4)) { LCDML.BT_left();  }
      if (bitRead(but_stat, 5)) { LCDML.BT_right(); }
    }
  }
  
  if((millis() - g_LCDML_DISP_press_time) >= 200) {
    g_LCDML_DISP_press_time = millis(); // reset debounce timer

    uint16_t valuex = analogRead(_LCDML_CONTROL_analog_pinx);  // analogpinx
    uint16_t valuey = analogRead(_LCDML_CONTROL_analog_piny);  // analogpinx
    
    if (valuey >= _LCDML_CONTROL_analog_up_min && valuey <= _LCDML_CONTROL_analog_up_max) { LCDML.BT_up(); }        // up
    if (valuey >= _LCDML_CONTROL_analog_down_min && valuey <= _LCDML_CONTROL_analog_down_max) { LCDML.BT_down(); }    // down
    if (valuex >= _LCDML_CONTROL_analog_left_min && valuex <= _LCDML_CONTROL_analog_left_max) { LCDML.BT_left(); }     // left
    if (valuex >= _LCDML_CONTROL_analog_right_min && valuex <= _LCDML_CONTROL_analog_right_max) { LCDML.BT_right(); }    // right

    //if(valuee == true) {LCDML.BT_enter();}    // enter

    // back buttons have to be included as menu item
    // lock at the example LCDML_back_button
  }
}
// *********************************************************************
// ******************************* END *********************************
// *********************************************************************

Jomelo, 2020-05-04:
Hello, 
there are really much possible solution. When it works it is good ;-)

When you find a reproduceable freeze i can look if there is a bug.

closed 2020-05-04 by Jomelo

romton843, 2020-05-07:
Thanks again, I don't know what I did before but it's working great.