Jomelo/LCDMenuLib2 #63: CONTROL WITH ENCODER

totza2010, 2020-05-04:
CONTROL WITH ENCODER    BT_quit, BT_enter Work.
                                                BT_left, BT_down, BT_right, BT_up not work.
test from code https://github.com/PaulStoffregen/Encoder/tree/master/examples/Basic work.
[image]

Jomelo, 2020-05-04:
Hi, 
how have you connect the encoder ? With my encoder it works correct.

totza2010, 2020-05-04:
#define encoder_A_pin       35
#define encoder_B_pin       34
#define encoder_button_pin  36

i use esp32

Jomelo, 2020-05-05:
I read the ESP32 datasheet for you:
https://randomnerdtutorials.com/esp32-pinout-reference-gpios/

> Input only pins
> GPIOs 34 to 39 are GPIs – input only pins. These pins don’t have internal pull-ups or pull-down resistors. They can’t be used as outputs, so use these pins only as inputs:
> 
> GPIO 34
> GPIO 35
> GPIO 36
> GPIO 39

You have to add extern pullup resistors.

Jomelo, 2020-05-17:
Is this problem fixed with extern pullups ?

closed 2020-05-26 by Jomelo