Jomelo/LCDMenuLib2 #64: goToRoot instead of Screensaver

nwa2001, 2020-05-18:
Hey, nice Library, thank you for your work!

I guess in the new version v2.2.3, there is the same Bug like in issue #31.

I have reconstructed the problem by using the example code (LCDML_u8g2lib):

In v2.2.3 you have to allocate the screensaver function to a concrete Menu Item, otherwise istead of showing the screensaver it will jump to the root menu, after the given time (default: 10seconds).

For testing i tried with v 2.1.2 .... the screensaver works as it should.

Thank you in advance.

Jomelo, 2020-05-19:
Hey, 
something is not clear to me.
On my tests i check the screensaver with a menu function and it works. Have you changed something on the example code ? 
Do you use the screensaver with a function which is not defined in the menu structure ?

I have add some defined states in this version. A defined state is the "goRoot" way. 
For example when a function is used which is not defined in the menu structure the menu goes to the root position.

nwa2001, 2020-05-19:
Hey..

_On my tests i check the screensaver with a menu function and it works._

Yes, as long as you call the screensaver function in one concrete Menuitem. Lets say you switch in the example:

  LCDML_add         (0  , LCDML_0         , 1  , "Information"      , **mFunc_information**);       // this menu function can be found on "LCDML_display_menuFunction" tab

to:

  LCDML_add         (0  , LCDML_0         , 1  , "Information"      , **mFunc_screensaver**);       // this menu function can be found on "LCDML_display_menuFunction" tab

then the screensaver works as it should.
Even without calling the (new) Menuitem1 a single time.

_Do you use the screensaver with a function which is not defined in the menu structure ?_

I didn't change anything, except the line for my display:

>     U8G2_ST7920_128X64_F_SW_SPI u8g2(U8G2_R0, 8, 6, 7, U8X8_PIN_NONE);

and the three encoder pins...

Greetings

nwa2001, 2020-05-19:
... something causes not to open screensaver funktion after the elapsed time ... maybe (but that's oracle reading) a function that is hidden is not registered by the code (sounds weird, but i'm not an expert)
I didn't even got a serial print (that I placed in the screensaver setup).

Jomelo, 2020-05-22:
Hello, 
i found something, but the changes are not so easy. I have to change some more lines.

Jomelo, 2020-05-23:
Can you test the latest master branch version ?

nwa2001, 2020-05-26:
Thank you Nils, the problem seems to be solved!
Nice!