Jomelo/LCDMenuLib2 #65: goBackToMenu(e) not working, always jumping to Root

nwa2001, 2020-05-26:
Hey Nils,

again thank you for your great work!

Maybe here is something not working as it should:

You say in the wiki:

> LCDML.FUNC_goBackToMenu(uint8_t e=0);   // no return / e=0 close function, e=1 close function and go a layer back

But in any case, when closing the screensaver you will end in root Menu!

This is what i testet in the screensaver function:

```
if(LCDML.FUNC_loop())           // ****** LOOP *********
  {
    if (LCDML.BT_checkAny()) // check if any button is pressed (enter, up, down, left, right)
    {
      //LCDML.FUNC_goBackToMenu();  // leave this function
      //LCDML.FUNC_goBackToMenu(0);  // leave this function
      LCDML.FUNC_goBackToMenu(1);  // leave this function
    }
  }

  if(LCDML.FUNC_close())          // ****** STABLE END *********
  {
    // The screensaver go to the root menu
    //LCDML.MENU_goRoot();
    //LCDML.FUNC_goBackToMenu();  // leave this function
    //LCDML.FUNC_goBackToMenu(0);  // leave this function
    LCDML.FUNC_goBackToMenu(1);  // leave this function
  }
}
```

as you see, i have tried those three ways, in any case i will end up in the root Menu. 
And i was deep in the Menuthree (Programm-Programm1-P1

Jomelo, 2020-05-26:
Hello, 
this is not a bug it is a feature ... ;)

This is a handling for hidden functions like the screensaver. 

**Why it is so:**
When the program leave a hidden function the cursor cannot be set to the correct position. The possition before the element is wrong, the possition after the element is wrong. Some peaple thing after an element is good other think before an element is good. I cannot handle this correct, for this problem i add a "go back advanced" configuration. This can be enabled in the function setup. Look on the following code example.

```
// *********************************************************************
void your_function_name(uint8_t param)
// *********************************************************************
{
  if(LCDML.FUNC_setup())          // ****** SETUP *********
  {
    // remmove compiler warnings when the param variable is not used:
    //LCDML_UNUSED(param);
    // setup
    // is called only if it is started

    // starts a trigger event for the loop function every 100 milliseconds
    LCDML.FUNC_setLoopInterval(100);

    // uncomment this line when the menu should go back to the last called position
    // this could be a cursor position or an active menu function
    // GBA means => go back advanced
    LCDML.FUNC_setGBA() 

    //
  }

  if(LCDML.FUNC_loop())           // ****** LOOP *********
  {
    // loop
    // is called when it is triggered
    // - with LCDML_DISP_triggerMenu( milliseconds )
    // - with every button or event status change

    // uncomment this line when the screensaver should not be called when this function is running
    // reset screensaver timer
    //LCDML.SCREEN_resetTimer();

    // check if any button is pressed (enter, up, down, left, right)
    if(LCDML.BT_checkAny()) {
      LCDML.FUNC_goBackToMenu();
    }
  }

  if(LCDML.FUNC_close())      // ****** STABLE END *********
  {
    // loop end
    // you can here reset some global vars or delete it
    // this function is always called when the functions ends.
    // this means when you are calling a jumpTo ore a goRoot function
    // that this part is called before a function is closed
  }
}
```

The GBA handling can be configured with three methods:

`LCDML.FUNC_setGBA();   // decide automatically (call a function when a function was active before, or call the menu position when the menu was active before`

`LCDML.FUNC_setGBAToLastCursorPos();`

`LCDM.FUNC_setGBAToLastFunc();`

The GBA functions have another reason to exists.  Whenever a function is called not from a menu function (for example by time or by an event or an extern jumpTo Handling which is not called from the menu function), it is good when the old state can be reconstructed.  

I hope this helps you. 

Please do not use the ..GBA.. function in combination with a goRoot on FUNC_close.  goRoot wins this duell. ;)

nwa2001, 2020-05-26:
OK, I didn't recognized the new function (was not available in 2.2.1)

_LCDML.FUNC_setGBA()_ 

only read in the wiki:
_LCDML.FUNC_goBackToMenu(uint8_t e=0);   // no return / e=0 close function, e=1 close function and go a layer back_    

my fault! (would you please put the setGBA handling in the wiki?)

It works nice for me, when:
- activating _LCDML.FUNC_setGBA()_ in the setup from screensaver
- let the button check part as it is :       _LCDML.FUNC_goBackToMenu();  // leave this function_
- comment in the closing part: _//LCDML.MENU_goRoot();_

=> jumps back to the last menu before the screensaver was activated

Thank you again!

Jomelo, 2020-05-26:
Hi, 
i have update the wiki page.

closed 2020-05-26 by Jomelo