Jomelo/LCDMenuLib2 #67: Dynamic titles of menus

Scabattoir, 2020-06-09:
Hi Jomelo,

first of all I'd like to thank you for such a nice working library. I did test some of the others but yours convinced me to implement it.

And now my issue:
I'd like to make a title kind of dynamic in a way as to make it up of a string and the value of a variable.
I was trying to make a second string variable adding up the first one and the variable but it seems like I cannot refer to a variable in the name field. It stops with the error of "cannot convert 'String' to 'char' in initialization
             char g_LCDML_DISP_lang_ ## lang ## _ ## name ##_var[] = {content}"

Would there be a way to display what I'm after?
Thank you,
Csaba

Scabattoir, 2020-06-10:
Does anyone have any ideas of how could I solve this please?

Scabattoir, 2020-06-12:
no way to do this then I guess, right?
Thanks

Jomelo, 2020-06-12:
Hello,
Sorry for my late answer. Please have a look on the dynamic menu elements on the examples. 
This elements can have dynamic content like parameter or a running time. 

Your way is not supported.

Scabattoir, 2020-06-13:
No worries, thank you for your answer.
I was asking about having a float (or int/10) displayed in the dynamic content but I couldn't find a way nor an example I could follow.
If I set the display variable of the dynamic content to float to display the int/10 inline then the app hangs on the teensy.
If you could guide me in the right direction I'd be very happy.
Thank you

Jomelo, 2020-06-13:
I can create example code, but at the moment i have no access to a latop or pc and on my handy i cannot write code. You have to wait some days.

Scabattoir, 2020-06-13:
Thank you Nils for your kindness, looking forward to seeing it.
In the meantime I have figured out the sprintf(); and the formats it can handle but still struggling with the dynamic line. I got as far as I can force it to display 0.0 but it won't work and to keep it working I can't make it accept anything else but integers.
Take your time though, I'll do a workaround until you find the time to light your knowledge and wisdom on me.
Best wishes,
Csaba

Jomelo, 2020-06-20:
Example 1: Dyn Function (simple Output)

```
float g_dynParam = 0.1; // when this value comes from an EEPROM, load it in setup
                          // at the moment here is no setup function (To-Do)
void mDyn_para(uint8_t line)
// *********************************************************************
{
  // check if this function is active (cursor stands on this line)
  if (line == LCDML.MENU_getCursorPos())
  {
    // make only an action when the cursor stands on this menu item
    //check Button
    if(LCDML.BT_checkAny())
    {
      if(LCDML.BT_checkEnter())
      {
        // this function checks returns the scroll disable status (0 = menu scrolling enabled, 1 = menu scrolling disabled)
        if(LCDML.MENU_getScrollDisableStatus() == 0)
        {
          // disable the menu scroll function to catch the cursor on this point
          // now it is possible to work with BT_checkUp and BT_checkDown in this function
          // this function can only be called in a menu, not in a menu function
          LCDML.MENU_disScroll();
        }
        else
        {
          // enable the normal menu scroll function
          LCDML.MENU_enScroll();
        }

        // do something
        // ...
      }

      // This check have only an effect when MENU_disScroll is set
      if(LCDML.BT_checkUp())
      {
        if(g_dynParam < 5.0)
        {
          g_dynParam += 0.1;
        }
      }

      // This check have only an effect when MENU_disScroll is set
      if(LCDML.BT_checkDown())
      {
        if(g_dynParam > 0.0)
        {
          g_dynParam -= 0.1;
        }
      }

      if(LCDML.BT_checkLeft())
      {
        if(g_dynParam > 0.0)
        {
          g_dynParam -= 0.1;
        }
      }
      if(LCDML.BT_checkRight())
      {        
        if(g_dynParam < 5.0)
        {
          g_dynParam += 0.1;
        }
      }
    }
  }

  Serial.print("dynValue:");
  Serial.print(g_dynParam);
}
```

Example 2: Dyn Function (with splitted output value)

```
float g_dynParam = 0.1; // when this value comes from an EEPROM, load it in setup
                          // at the moment here is no setup function (To-Do)
void mDyn_para(uint8_t line)
// *********************************************************************
{
  // check if this function is active (cursor stands on this line)
  if (line == LCDML.MENU_getCursorPos())
  {
    // make only an action when the cursor stands on this menu item
    //check Button
    if(LCDML.BT_checkAny())
    {
      if(LCDML.BT_checkEnter())
      {
        // this function checks returns the scroll disable status (0 = menu scrolling enabled, 1 = menu scrolling disabled)
        if(LCDML.MENU_getScrollDisableStatus() == 0)
        {
          // disable the menu scroll function to catch the cursor on this point
          // now it is possible to work with BT_checkUp and BT_checkDown in this function
          // this function can only be called in a menu, not in a menu function
          LCDML.MENU_disScroll();
        }
        else
        {
          // enable the normal menu scroll function
          LCDML.MENU_enScroll();
        }

        // do something
        // ...
      }

      // This check have only an effect when MENU_disScroll is set
      if(LCDML.BT_checkUp() || LCDML.BT_checkRight())
      {
        if(g_dynParam < 5.0)
        {
          g_dynParam += 0.1;
        }
      }

      // This check have only an effect when MENU_disScroll is set
      if(LCDML.BT_checkDown() || LCDML.BT_checkLeft())
      {
        if(g_dynParam > 0.0)
        {
          g_dynParam -= 0.1;
        }
      }     
      
    }
  }

  Serial.print("dynValue: ");
  // output value before the comma 
  Serial.print((int)g_dynParam);
  // ouput of the comma
  Serial.print(".");
  // ouput behind the comma
  Serial.print(((int)(g_dynParam*10))%10);  
}

```

closed 2020-11-30 by Jomelo