Jomelo/LCDMenuLib2 #69: Compatibility between version 2.2.2 and the latest

riquebrenner, 2020-10-29:
Greetings. 

Dear developer, I use your code to display information on a 4x20 display connected to an I2C bus. 

However, since version 2.2.3 the screensave is no longer called at the specified time. 

I made a diff with the example and I find no differences that can determine the problem. 

I also see that the determined timeout is executed the display flashes and returns to the main menu. 

I use the example LCDML_i2c_dislpay_20x4 as a reference. Am I doing something wrong?

Best regards,
Henrique Aschenbrenner

Jomelo, 2020-11-11:
Hi, will check it soon. At the moment i have no i2c display here.

Jomelo, 2020-11-28:
Hello, 
today i have checked the library with an i2c 20x4 display. I could not found any bug. But i seen that the version 2.2.3 is not the newest version. Could you check your program with v2.2.6 ?

Changelog:
https://github.com/Jomelo/LCDMenuLib2/releases

Best Regards

closed 2022-12-30 by Jomelo