Jomelo/LCDMenuLib2 #71: lcdmlib2 on ttgo lora32 oled v1 and modbus slave rtu

loscorpione, 2020-11-26:
Hi Nils
 I used his library previously and now I'm using version 2 on module ttgo lora32 oled v1
I had to make a modbus RTU slave and for that purpose I'm using Alexander Emelianov's library 
[https://github.com/emelianov/modbus-esp8266](url)
I tested the library with the attached example:

> #include <ModbusRTU.h>
#define REGN 10
#define SLAVE_ID 1
ModbusRTU mb;
void setup() {
  Serial.begin(9600, SERIAL_8N1);
#if defined(ESP32) || defined(ESP8266)
  mb.begin(&Serial);
#else
  mb.begin(&Serial);
  mb.setBaudrate(9600);
#endif
  mb.slave(SLAVE_ID);
  mb.addHreg(REGN);
  mb.Hreg(REGN, 100);
}
void loop() {
  mb.task();
  yield();
}

the library works perfectly but if I insert it in the scketch with the lcdmlib2 I have continuous restarts of the program.
I found that this happens by putting this part in the void loop

> void loop() {
    // this function must called here, do not delete it
    LCDML.loop();
   mb.task();
   yield();
} 

I suppose I'm entering
>    mb.task();
   yield();

in the wrong place
Could you show me the correct way?
i don't know how to fix this
congratulations for the great work you have done with this library

ciao !!!

Jomelo, 2020-11-28:
Hello,
you need here a systematic troubleshooting for this restart. Please check the following things:
- is the power supply sufficient for the selected hardware?
- drives one pin more current as allowed ?
- how many dynamic memory is used ?
- can it happen that one part of the software needs to much times (search after  "Disable esp8266 watchdog reset") ?

ciao

loscorpione, 2020-12-01:
- is the power supply sufficient for the selected hardware? **yes I measured it**
- drives one pin more current as allowed ?**for the moment the only pins are connected to the uart-rs485 module and 4 buttons**
- how many dynamic memory is used ? **5% of the dynamic memory is used** 
- can it happen that one part of the software needs to much time (search after "Disable esp8266 watchdog reset") ?**This could be but how can I fix it?**
- i do not know much over the modbus library, is there any interrupt based handling used ?**I am not aware that the library uses interrupts**

Thanks for your interest

closed 2022-12-30 by Jomelo