Jomelo/LCDMenuLib2 #74: ST7735 display is flickering

maxholgasson, 2021-02-02:
Hi,
I tested your example LCDML_adafruit_gfx_st7735 and noticed some weird flickering when I click through the menu.
I only used SSD1306 displays so far and the operation is very smooth with those
Is there a way to fix it? MCU is a SAMD21

https://user-images.githubusercontent.com/4289347/106612451-66775480-6569-11eb-9b1f-346417f10b21.mp4

Jomelo, 2021-02-22:
Hi, 
i have fix a bug today. The display was refreshed with every cursor movement.

Can you test the lates beta version v2.2.7 beta 2.

Best Regards

rsp-git, 2021-09-03:
Where can i find the beta2? I Use the 2.2.7 release and there is the same problem.
The Refresh let's the display flickr.
In the INO Files inside your Project should be the file LCDML_display_menu.ino and there is the "refresh" Function.
I added a if section arround fillscreen command:
`  if (LCDML.DISP_checkMenuUpdate()) {
      display.fillScreen(_LCDML_ADAFRUIT_BACKGROUND_COLOR);
  }`

Then I switch from X as cursor to Color as Cursor - before i only draw a black box over the X positions:
`if(tmp->checkType_menu() == true)
        {
            LCDML_getContent(content_text, tmp->getID());
            display.setCursor(0, _LCDML_ADAFRUIT_FONT_H * (n));
            if(LCDML.MENU_getCursorPos()==n)
              display.setTextColor(_LCDML_ADAFRUIT_HIGHLIGHT_COLOR); 
            else
              display.setTextColor(_LCDML_ADAFRUIT_TEXT_COLOR); 
            display.println(content_text);
          /*if (LCDML.DISP_checkMenuCursorUpdate()) {
          } else {
            display.fillRect(0, _LCDML_ADAFRUIT_FONT_H * (n), 16,  _LCDML_ADAFRUIT_FONT_H, _LCDML_ADAFRUIT_BACKGROUND_COLOR);
          }*/          
        }`
Everything works fine - only for the dynamic value. there i need to know how to detect this. If the function lcdml_menu_display() is called it clears the display, also if there is no "menu update". How can i detect this? Then everything works fine.

@maxholgasson: Do you use interrupts for encoder? Tell me how :)

Jomelo, 2022-12-30:
Hi, 
i will check this in the next weeks.

localhero44, 2023-11-06:
Hi guys, is this problem solved ? I also notice flickering.

Jomelo, 2023-11-14:
When this bug also happens this is not fixed. 
I have to order a new display to test it. All my displayes are build in some projects. 

I will check it.

Jomelo, 2025-04-21:
I have reproduce the bug on my small OLED screen. I´m looking for an bug fix in the next weeks.

indalg, 2025-04-22:
Gracias