Jomelo/LCDMenuLib2 #77: PlatfromIO : compatible example

AvinashPudale, 2021-05-30:
I just want to use vs code with platform io for compile and upload code .but  current multi tab. ino examples not work  .
 when imported serial example at vs code  code compile upload to esp32.  but not work . I just open  example folder in using import Arduino project and added library needed for project.

Jomelo, 2021-05-30:
Hello, 
i will try to use platform io with vscode. At the moment i am not using this constellation.

Jomelo, 2021-06-08:
Hi, 
i have test it, but i could not compile any arduino program correctly. The bad think was, that after the installtion of  plattform io my python implementation does not work correctly. I think i have generally chose the wrong way to install it. 
At the moment i have no answer to your question.

AvinashPudale, 2021-06-09:
Thanks for looking into this.  Hoping it will resolve soon.  😀.

Jomelo, 2022-12-30:
I add an example for plattform io to the master versioin

closed 2022-12-30 by Jomelo