Jomelo/LCDMenuLib2 #78: analogue joystick's button

youxiaojie, 2021-07-03:
// *********************************************************************
// *************** (6) CONTROL OVER JOYSTICK *********************
```
// *********************************************************************
#elif(_LCDML_CONTROL_cfg == 6)
	// settings
	#define _LCDML_CONTROL_analog_pinx A0
	#define _LCDML_CONTROL_analog_piny A1
	#define _LCDML_CONTROL_digitalread 33 //don't work with u8glib

	// when you did not use a button set the value to zero
	#define _LCDML_CONTROL_analog_up_min 612 // Button Up
	#define _LCDML_CONTROL_analog_up_max 1023
```
u8g2 does not support joystick's button, how I can do?

Jomelo, 2021-07-03:
The control method have nothing to do with the display library. 

With a joystick i mean this components:
https://create.arduino.cc/projecthub/MinukaThesathYapa/arduino-thumb-joystick-to-processing-92c182

youxiaojie, 2021-07-05:
#define _LCDML_CONTROL_digitalread 33 //don't work with u8glib

I see the comment //dont work with u8glib, I feel strange...

closed 2022-12-30 by Jomelo