Jomelo/LCDMenuLib2 #80: Экспро проекта в VSC

AlexArGC, 2021-07-28:
Good afternoon. I'm just learning C ++.
Why doesn't the example compile? The VSC compiler (1.58.2) swears at lines like:
`LCDML_add         (0  , LCDML_0         , 1  , "Mode"             , mFunc_mode);`

no instance of constructor "LCDMenuLib2_menu::LCDMenuLib2_menu" matches the argument list -- argument types are: (int, int, int, void (), int)C/C++(289)

And also swears at:
`LCDML.SCREEN_enable(mFunc_screensaver, 10000); // set to 10 seconds`

argument of type "void (*)()" is incompatible with parameter of type "LCDML_FuncPtr_pu8"C/C++(167)

Jomelo, 2021-08-01:
Hello, 
are you using the arduino IDE ? On my current version 1.8.12 and 1.8.15 there are no errors. 
I think when you are using another editor the includes are in the wrong order.

AlexArGC, 2021-08-15:
I am sorry for disturbing you. Understood. Adapted VSC + PlatformIO code. Everything worked out when I used the header files.

rsp-git, 2021-09-03:
> I am sorry for disturbing you. Understood :-) I tried to adapt the VSC + PlatformIO code. Everything worked out when I fixed the header files. Thank you so much for such a great library!

Can you tell me what you've done? I have a simillar issue.

AlexArGC, 2021-09-03:
> 
> 
> > I am sorry for disturbing you. Understood :-) I tried to adapt the VSC + PlatformIO code. Everything worked out when I fixed the header files. Thank you so much for such a great library!
> 
> Can you tell me what you've done? I have a simillar issue.

The easiest way is to combine the code into a * .h and * .cpp file.

Jomelo, 2022-12-30:
I add a plattform io example to the master version. I test it with VSC.

closed 2022-12-30 by Jomelo