Jomelo/LCDMenuLib2 #81: pullup even not defined

specs32, 2021-08-29:
Hi, first of all thanks for your nice piece of software ! 

I use the ST77XX example to build a new sketch on an ESP32S2.

In the main sketch I have defined a PIN like this :
`#define ENBST 33`
in the setup of my main sketch I use :
`  pinMode(ENBST, OUTPUT);           // boost  function`
`  digitalWrite(ENBST, LOW);         // off by default`
to pull it LOW.

When I flash the Sketch the PIN 33  is always HIGH, even it is not defined or talked to anywhere else in the sketch and all other files as far as I can see. (even searched all the source for "33")

I can use the same sketch without your lib attached and the PIN gets pulled LOW and stays that way.

Whats going on ? 
Thanks and greetings.

Jomelo, 2021-08-29:
Hello, 
the Lib use internal no pinMode / digitalWrite functions.
When there is something wrong it could only be one of the other libs or some code in the examples like for controling the menu. Are you are using further libs ?

Best Regards,

specs32, 2021-08-29:
Hi, thanks for your quick answer on a Sunday afternoon :D !!
I just compiled the demo sketch for the st7735 WITHOUT any of my code, I have  selected "digital pins" (2) as input method and set the  values for my three buttons accordingly (37, 38, 39, ) I have the same issue that pin 33 gets pulled high.... I then added the definition for pin 33 and pulled it low in setup, but it gets pulled high again right after boot. None of my sketch was involved in this test.

closed 2022-12-30 by Jomelo