Jomelo/LCDMenuLib2 #82: PlatformIO Port

rsp-git, 2021-09-03:
Hello,
i have tried to run my Code from [ArduinoIDE](https://github.com/rsp-git/Color-Display-LCDML) in PlatformIO.
In Arduino IDE it works - PlatformIO can't compile it.
I've created headerfiles for everything to convert .ino to cpp Style.
This is my current Testversion: https://github.com/rsp-git/Color-Display-LCDML_PlatformIO

I don't know how to fix it:
`src\LCDML_display_menu.cpp: In function 'void lcdml_menu_display()':
.pio\libdeps\mzeroUSB\LCDMenuLib2\src/LCDMenuLib2_macros.h:103:54: error: 'g_LCDML_DISP_lang_lcdml_table' was not declared in this scope
                 strcpy_P(var, (char*)pgm_read_word(&(g_LCDML_DISP_lang_ ## lang ## _table[id])));`

Can you help me?
Do i need pgm_* Stuff in SamD21 Processors? I think this can also run from Memory - or?

BR Raphael

larryl79, 2022-05-09:
> Hello, i have tried to run my Code from [ArduinoIDE](https://github.com/rsp-git/Color-Display-LCDML) in PlatformIO. In Arduino IDE it works - PlatformIO can't compile it. I've created headerfiles [...]

I made a solution for it.

in a LCDMenuLib2.h I've added  in a line 42
// this line need for resolve vsCode & platformio lib compile error may resolves another platforms too
    extern char* g_LCDML_DISP_lang_lcdml_table[254];

so may you need pull the latest repo from github if you haven't got that lines in that file

Jomelo, 2022-12-30:
I have add a plattform io example to the master version.

closed 2022-12-30 by Jomelo