Jomelo/LCDMenuLib2 #92: Problem with "extern char* g_LCDML_DISP_lang_lcdml_table [254]" on Arduino -IDE (2.03)

dcoredump, 2022-12-17:
Hi,

I am getting problem with the extern declaration for `g_LCDML_DISP_lang_lcdml_table` in `LCDMenuLib2.h:43`:

`/home/******/Arduino-Teensy/libraries/LCDMenuLib2/src/LCDMenuLib2.h:43:18: note: previous declaration as 'char* g_LCDML_DISP_lang_lcdml_table [254]'`

The best fix for this would be to disable this declaration when using Arduino-IDE - but I currently have no idea how to check this.

Regards, Holger

piotrbod111, 2022-12-20:
Hello, 
Have probably similar problem, platformIO IDE, Arduino device. 

"conflicting declaration 'const char* const g_LCDML_DISP_lang_lcdml_table []'"
or other random errors in LCDMenuLib2_macros.h file after clean build. 

Resently added Extern in LCDMenuLib2.h do not change anyting.

Any idea ? 

Reagars 
Peter

Jomelo, 2022-12-30:
Hi, 

@dcoredump 
In LCDMenuLib2 versions i have not add this change because it is difficult. Plattform IO needs this line but with this line the arduino IDE does not compile. Without this line Plattform IO does not work. I have searched for a special definition wich is only set when plattform io is used. But i have nothing found. 
I could try to use this:
   #ifndef ARDUINO
      include ...
   #endif
But i do not know if it works corectly. When arduino.h is implemented before lcdmenulib is included it does not work. Only when i change the example code to define a special variable i can handle this, but i need differnt example ways for plattform io and arduino ide ... 

This pullrequest adds this line: https://github.com/Jomelo/LCDMenuLib2/pull/86

@piotrbod111 
Please use the newest master version. I add an plattform io example. 
Please edit the LCDMenuLib2.h on line 47 and remove the comment symbol  -> #define _LCDML_plattform_io_support 

Best regards,
Nils

tusker-tools, 2023-02-26:
Hello,
I'm trying to port this lib to use it with STM32CubeIDE for an STM32F103 (Blue Pill). Even with the mentioned `#define _LCDML_plattform_io_support` I get a compiler error (Compiler g++ 2021.10):
```
./Drivers/Display/LCDMenuLib2/src/LCDMenuLib2_macros.h:104:32: error: conflicting declaration 'const char* const g_LCDML_DISP_lang_lcdml_table []'
  104 |             const char * const g_LCDML_DISP_lang_ ## lang ## _table[] PROGMEM = { LCDML_DISP_lang_repeat(N, lang) }
      |                                ^~~~~~~~~~~~~~~~~~
../Drivers/Display/LCDMenuLib2/src/LCDMenuLib2_macros.h:135:9: note: in expansion of macro 'LCDML_createCustomLang'
  135 |         LCDML_createCustomLang(N, lcdml)
      |         ^~~~~~~~~~~~~~~~~~~~~~
../Drivers/Display/LCDMenuLib2/src/menu.cpp:52:1: note: in expansion of macro 'LCDML_createMenu'
   52 | LCDML_createMenu(_LCDML_DISP_cnt);
      | ^~~~~~~~~~~~~~~~
In file included from ../Drivers/Display/LCDMenuLib2/src/menu.cpp:9:
../Drivers/Display/LCDMenuLib2/src/LCDMenuLib2.h:72:16: note: previous declaration as 'char* g_LCDML_DISP_lang_lcdml_table [254]'
   72 |   extern char* g_LCDML_DISP_lang_lcdml_table[254];
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

In my opinion this is because we defined the related array as "const char * const" as shown below:
```
    #define LCDML_createCustomLang(N, lang) \
            const char * const g_LCDML_DISP_lang_ ## lang ## _table[] PROGMEM = { LCDML_DISP_lang_repeat(N, lang) }
```
But in the extern declaration  `extern char* g_LCDML_DISP_lang_lcdml_table[254];` , which was added in #86, it is now "char * const", which does not match the definition.

Could somebody pls. confirm?

Regards,
Philipp

piotrbod111, 2023-02-27:
@tusker-tools   thesame errors, platfomio, Arduino.

tusker-tools, 2023-03-05:
I was able to solve the compiler error by changing the mentioned extern declaration in LCDMenuLib2.h as below:

```
    #ifdef _LCDML_plattform_io_support
        // this line need for resolve vsCode & platformio lib compile error may resolves another platforms too
		extern const char * const g_LCDML_DISP_lang_lcdml_table[254];
    #endif
```