Jomelo/LCDMenuLib2 #94: ESP32 + Encoder + LCD 20x4 funktioniert nicht

Axl1978, 2023-09-20:
Hallo! 
Brauche Hilfe , schaffe es einfach nicht einen Encoder mit LCDMenuLib2 auf einen ESP32 zum laufen zu bekommen.
Bin schon mal draufgekommen des die Encoder.h an ESP32 nicht funktioniert.

Dann hab ich diesen Code von [khvalera](https://github.com/khvalera)  hier https://github.com/Jomelo/LCDMenuLib2/issues/89 gefunden und dachte das wars jetzt.
Leider funktioniert bei mir nur der Button.

Kann mir bitte jemand helfen?
Nehme ich vielleicht die falschen GPIO's?
Hab einen Encoder der auf einer Platine verbaut ist. Hat auf der Rückseite 3 kleine Widerstände. Aber auch auslöten ändert nichts.

Danke

Jomelo, 2023-09-21:
Moin, 
teste doch erstmal eine Encoder Library ohne die LCDMenu2 Lib. Stell gerne deinen Code zur Verfügung und beschreibe wie du den Encoder mit dem Controller verbunden hast. Gibt es irgendwelche Widerstände ? Werden interne Pullups oder Downs genutzt ?

Zur Verkabelung (Wiring) gibt es viele Beispiele. 
[](https://www.google.com/search?sca_esv=567294360&sxsrf=AM9HkKl3pYL3teI4ePYCHbSNs20myUFkzw:1695318015890&q=arduino+encoder+wiring+esp32&tbm=isch&source=lnms&sa=X&ved=2ahUKEwjZ7_jgn7yBAxUE3gIHHTv0AWAQ0pQJegQIBxAB&biw=2284&bih=1167&dpr=1.1#imgrc=LVxu9duIeNPh8M)

Teil gerne mehr Informationen, dann bekommen wir das schon hin.

Axl1978, 2023-09-24:
Hallo! Ich danke dir für deine rasche Antwort. Hab die letzten Stunden etliche Encoder Library probiert und alle GPIO die es gibt.
Schlussendlich hab ich den Fehler gefunden!!

Es ist so peinlich es zu sagen, aber des Problem war ein Kabelbruch der zeitweise Kontakt hatte und dann wieder nicht.
Jetzt funktioniert alles tadellos.

Zu meinem Projekt:

Will für meinen Garten eine Lichtsteuerung aus 2x ESP32 bauen. 
Einer dient als Sender (mit LCD 20x4 mit Encoder) um verschiedene Lichtzweige einzeln oder als Gesamtszene an einen Empfänger zu senden.

Der Empfänger (auch mit LCD 20x4, Encoder und einem 16-Relais Modul 12V mit Optokoppler ) steuert dann das Modul an.
Zum Glück sind alle Lichtleitungen zentral zusammengefasst in der Garage.

Bin neu was die Programmierung mit ESP32 betrifft, aber werde mit der ESPNOW Bibliothek arbeiten.
Muss nur mehr herausfinden wie ich einzelne char oder int , bzw. eine Art Array senden kann.
Aber das werde ich schon schaffen.

Mit deinem Menü hab ich schon viel auf Arduino gemacht und komm sehr gut damit zurecht.
Steuerung einer Beleuchtung mit 5x WS2812B 5M, oder eine komplette Steuerung für ein Gewächshaus.

Danke auf jeden Fall für dein Engagement und deine Bereitschaft zur Hilfestellung.

LG Alex

closed 2023-09-25 by Jomelo