Jomelo/LCDMenuLib2 #99: Menu not displayed correctly when using SD Card

j0k3r187, 2024-02-16:
Hello,

Im trying to use this Library and an SD Card together, but when i Initialize SD library with SD.begin(SD_CS) 
in Setup the Menu not displayed correctly.

Im using an Arduino Mega with GLCD ST7920 on SPI Mode. LCD_CS and SD_CS are not the same!

If you try the Example LCDML_03021_u8g2lib and just include SD Lib and SD.begin(SD_CS) 
in Setup you can reproduce this.

Sometimes it looks good after Uploading but after Arduino Reset its the same Error again.

Jomelo, 2024-02-17:
Hello, 
have you test the differnt libraries without the LCDMenuLib2 ? 
The Library is not using external library content into the lib. The external libs are only used in the examples. 

Please test the SD Card library with the u8g2lib first for compatibility.
Best regards.

j0k3r187, 2024-02-18:
Hello,

i tested u8g2 and SD on HW_SPI with my ST7920_128x64 GLCD and it seems working. Didnt do deep testing with u8g2 and SD
but it looks like the bug only appears with your Library.
Dont get me wrong, you have done a great job with this lib and thanks for creating this.
Maybe you can check if you getr the same Problems with ST7920_128x64 GLCD, u8g2 and SD on HW_SPI.

Otherwise everybody who have the same problem, i solved mine by seperating the SPI Communication.
Im using now HW_SPI on SD-Card Reader and a SW_SPI on the Display.

Hope this helps and thanks again for this lib :)

closed 2024-04-14 by Jomelo