Jomelo/LCDMenuLib2 #100: Help

dimontau, 2024-03-31:
Hello. Probably a stupid question, but I can't figure out how to use this library. I use a 128X64 display, an encoder and an htu21 sensor. I have readings from the sensor displayed. How can I bring up the menu from the main display and then return from the menu to the main display?
Sincerely.
Thanks for the help.

Jomelo, 2024-04-01:
Hello, 
have you tryed the examples ? 
On Arduino IDE:  Press File-> Examples -> LCDMenuLib2 -> 03_displaytzpes -> 02_glcd -> LCDML_03021_u8g2lib

https://github.com/Jomelo/LCDMenuLib2/tree/master/examples/03_displaytypes/02_glcd/LCDML_03021_u8g2lib

When you are testing the example, go to LCDML_control.ino Tab and set 

#define _LCDML_CONTROL_cfg      3

Scroll after this to line 244 and make your configuration settings. 

When you are using an esp you have to set the configuration to 8 and scroll to the end of the file for the settings.

dimontau, 2024-04-02:
Yes. I figured it out. My main screen displays temperature and humidity. I can't figure out how to bring up the menu and then return from the menu to the main screen with temperature and humidity. Is it possible to have an example of calling the menu from the main screen and returning from the menu back to the main screen?

Jomelo, 2024-04-02:
Can you use a menu function as main screen. It will be the fastes implementation. You can start the menu in a menu function as main screen when you add a "jumpToFunc" function in the setup of the programm.

closed 2024-04-14 by Jomelo