Jomelo/LCDMenuLib2 #101: Wrong memory access - Menu items show weird symbols

tusker-tools, 2025-01-08:
Hello,
when using the library on STM32 I encountered a bug in the display of the menu items.

This is due to the following reason:
My g_LCDML_DISP_lang_lcdml_table[0] holds the address 0x8012bcb (this is the address where string of menu item id=0 is starting).

LCDML code is like below:

```
/* Menu Main Function */
LCDML_getContent(content_text, tmp->getID());

/* LCDMenuLib2_macros.h */
#define LCDML_getContent(var, id) \
    LCDML_getCustomContent(lcdml, var, id)
...
#define LCDML_getCustomContent(lang, var, id) \
      if(id < _LCDML_NO_FUNC) {\
          strcpy_P(var, (char*)pgm_read_word(&(g_LCDML_DISP_lang_ ## lang ## _table[id]))); \
      } 
```
By using pgm_read_word, only 16 bits of the 32-bit address is read. So the address is truncated from 0x8012bcb to 0x2bcb. Subsequently, the text string of menu item id=0 is read from a totally wrong address, causing a display of completely wrong symbols.

Proposal:
From my point of view, this bug effects all target plattforms with 32bit address space. It should urgently be fixed. For instance, pgm_read_word could be replaced by pgm_read_dword.

Please tell my your opinion.

Best regards,
Philipp

Jomelo, 2025-01-11:
Hi Philipp, 
thank you for reporting this. I will fix it in the next days. 

Best regards, 
Nils

Jomelo, 2025-04-21:
Hi Philipp, 
the bug is fixed. 
Best regards, 
Nils

closed 2025-04-21 by Jomelo