Jomelo/LCDMenuLib2 #102: UNO R4

Gehaktbalbugger, 2025-04-13:
Please provide a solution for the UNO R4. The library doesn't work on that CPU

Jomelo, 2025-04-13:
Hi, 
thank you for reporting. I will check this over Easter.
Best regards, 
Jomelo

Jomelo, 2025-04-21:
Hello, 
that is the same bug like #101 . The new Board is using a 32 bit CPU. 

As workaround you can uncomment the following line in the source code of the examples.

// special settings
// *********************************************************************
// enable this line when you are not usigng a standard arduino
// for example when your chip is an ESP or a STM or SAM or something else

**//#define _LCDML_cfg_use_ram** 

To fix the bug I need to find a define which I can include in the PROGMEM macros. With the define it is then possible to differentiate between <32 bit CPUs and 32 bit CPUs. I try to fix it in the next days.

Jomelo, 2025-04-21:
The bug is fixed in the latest file base.

closed 2025-04-21 by Jomelo